paukstelis/Octoprint-Cancelobject #1: Not working

nicozorza, 2018-06-01:
Good afternoon. I recentely tested your code but I couldn't make it work. I used Simplify3D to create two dummy pieces in two differente processes, but when I cancelled one of the processes, both of them kept printing.
I'm currentely using OctoPrint 1.3.8 and Simplify3D 4.0.1.
Thank you in advance.
Nicolás Zorzano.

paukstelis, 2018-06-05:
Please install the most recent version and provide your octoprint.log file for when the object is cancelled and when it hits a skipped object. Additionally, it would be helpful to provide the S3D factory file and/or gcode.

CNCKitchen, 2018-06-10:
Unfortunately it doesn't work for me either. The 3 processes are recognized and I can cancle them, but nothing happens. Also noticed that the statusbar always shows NONE:
[image]
Here are the settings (nothing altered):
[image]
And here a GCODE snippet:
[image]

CNCKitchen, 2018-06-10:
The lines seem to be replaced in the G-Code but still all the parts keep printing, when I cancel them:
[image]
I'd love to get this working!

paukstelis, 2018-06-10:
Can you provide your octoprint.log file? This will be helpful for debugging.

CNCKitchen, 2018-06-10:
Of course, here you go:
[octoprint.log](https://github.com/paukstelis/Octoprint-Cancelobject/files/2088009/octoprint.log)

paukstelis, 2018-06-10:
Thanks. This behavior seems odd since it should never reach that point with a None object. You might try changing the object replacement tag to something else or just retyping it and then saving (followed by restarting octoprint). Also, you may want to turn off your timelapse since it is generating errors that make it difficult to look through the log. Not sure if it matters, but can you provide some more details about the server running octoprint? RaspPi, etc.?

paukstelis, 2018-06-10:
Also, if you make the object replacement tag something else, you will need to re-upload the gcode file.

CNCKitchen, 2018-06-10:
Unfortunately still doesn't work. I attached another log (camera disabled ;-) )
[octoprint (2).log](https://github.com/paukstelis/Octoprint-Cancelobject/files/2088137/octoprint.2.log)

paukstelis, 2018-06-10:
Thanks again. It does seem that for whatever reason the main object list is getting lost. If you could, delete the plugin and install it from this URL: https://github.com/paukstelis/Octoprint-Cancelobject/archive/debugging.zip using the plugin manager. This just has some more logging information to see what is happening. You might also try connecting to the virtual printer included in octoprint so you don't actually have to keep starting prints!

CNCKitchen, 2018-06-10:
Here you go:
[octoprint (3).log](https://github.com/paukstelis/Octoprint-Cancelobject/files/2088172/octoprint.3.log)

paukstelis, 2018-06-10:
Great. I think I see the issue. Just need to figure out how to deal with it.

paukstelis, 2018-06-10:
Please reinstall the debugging branch again and see if that works. If not, please provide me your gcode so I can test locally.

paukstelis, 2018-06-10:
Or simpler yet, just change the Object Regular expression line to be: `; process (.*)\s`

CNCKitchen, 2018-06-10:
Unfortunately still doesn't work. Here is the log and the GCode file:
[octoprint (4).log](https://github.com/paukstelis/Octoprint-Cancelobject/files/2088209/octoprint.4.log)

[KingMarvin_5gcode.txt](https://github.com/paukstelis/Octoprint-Cancelobject/files/2088210/KingMarvin_5gcode.txt)
Thank you very much for your efforts!

paukstelis, 2018-06-10:
OK! I think I have it worked it. It was due to a carriage return being included  in the object name when the file was re-written. I have it working with your gcode on my end. Please reinstall debugging branch and test it there. I'll push this into the master if it works, but it highlights some other things I hadn't considered that I need to work on. Thank you very much for helping me work this out.

CNCKitchen, 2018-06-11:
Works now! Thank you very much for your support.

paukstelis, 2020-08-25:
@CNCKitchen made a nice youtube video long ago. I'm closing after 2+ years!

closed 2020-08-25 by paukstelis