paukstelis/Octoprint-Cancelobject #2: [question] what specifically does this plugin look for?

mer2329, 2018-06-09:
what exactly does this look for.
the reason i ask is slic3r prusa edition does support injecting code between objects 
its in gcode scripts, and is called "Between objects G-code (for sequential printing)"
now i use sequential printing. so im wondering what is specifically needed here. for this to work

paukstelis, 2018-06-09:
Right now it looks specifically for a comment (any line that starts with ";") then checks for matches to some regular expression that can be defined by the user (check the settings). The defaults is `; process (.*)` which is what S3D uses for every process switch. It is also important that the slicer resets the extrusion between objects (G92 E0).

paukstelis, 2018-06-10:
After looking through the source code I found something that will work for sequential printing in Slic3rPE.

For the start custom GCODE, include at the end: `; process 0`
For the end custom GCODE, include at the start: `; process ENDGCODE`
For the between object custom GCODE, include: `; process [current_object_idx]`

If you have four sequential objects, this should give you something like this:
[image]

Keep in mind, this is just to make it work with all the defaults of the plugin. You could of course change things around using regular expressions. Also, this doesn't take into account correcting for any retractions or extrusion resets which you might also need to do.

paukstelis, 2018-06-10:
I've updated  the readme to include some of this information. Let me know what you discover and I will update accordingly.

closed 2020-08-25 by paukstelis