paukstelis/Octoprint-Cancelobject #3: Layer change skipped

CNCKitchen, 2018-06-11:
After trying to figure out, why my prints didn't look nice, as soon as I canceled one part, I noticed that layer changes are skipped, if they are within one part. So usually one layer of a part is printed and then the next one is started before it moves to the next part, where the same process is repeated. If I now cancel one part, one layer change is skipped, so the extrusion is squished and then the following is two layers thick.

Any idea how to fix that? Can you avoid removing layer changes?

paukstelis, 2018-06-11:
If I understand correctly, the Z height is not correct after canceling? This may have been something I never noticed because I use Z-hop when retracting, so the Z height is set at the start of each object. I'll have to look into it further.

CNCKitchen, 2018-06-11:
Yes, a not correct z height after canceling sums it up and you are right,
z-hop works around that problem. That's a quick fix for me but I think it's
definitely worth looking into in the future.

paukstelis, 2018-06-11:
I appreciate the report. This is exactly the kind of testing I've needed to see (which is why I went ahead and released it). I will consider how best to handle it. It would be possible to parse all the skipped commands and let certain ones through.

paukstelis, 2018-06-11:
You can have a look at the allowed branch: https://github.com/paukstelis/Octoprint-Cancelobject/archive/allowed.zip

I haven't been able to test it yet. My main worry is that there will be a performance hit having to check each skipped line against the list of allowed  commands. Not sure how bad it will be, but maybe a python expert will weigh in if there is a more efficient way to do this.

paukstelis, 2018-06-11:
This looks like it works with the gcode you sent me in the last thread, but let me know how it goes when you get a chance to try it.

CNCKitchen, 2018-06-12:
Perfect! Problem solved! Though, as you mentioned, I have the impression, that it sometimes stutters a little.

paukstelis, 2018-06-12:
Great. What I think I'll do is make this something that can be enabled/disabled, depending on needs. Two things that will probably help with eliminating the stuttering will be removing the debugging print statement and decreasing the number of things in the list (probably don't need M204, M205, but I included them for testing).

closed 2018-06-17 by CNCKitchen