paukstelis/Octoprint-Cancelobject #5: Bigger job just stops at some point

CNCKitchen, 2018-06-16:
Have been playing around with the plugin for a bit and now have the problem that the print just stops at some point after canceling an object. Currently have no idea what the reason might be. This already happened twice yesterday on a job where I canceled a part after a while and today also already twice again on two other jobs after canceling. Added the LOG and the GCODE for debugging:
[octoprint (6).log](https://github.com/paukstelis/Octoprint-Cancelobject/files/2108030/octoprint.6.log)
[puzzle2.txt](https://github.com/paukstelis/Octoprint-Cancelobject/files/2108031/puzzle2.txt)

paukstelis, 2018-06-16:
Have you installed the most recent version of the master branch? 
From the logs it looks like the `#3dpart` lines are making it through to the printer.  
I'll look into it.

CNCKitchen, 2018-06-16:
No, I still used the debugging branch. I'll install the master and try it again. Thanks!

CNCKitchen, 2018-06-16:
Tried it again with the master branch, same behavior. Log file is attached.
Tired it again with ObjectCancel deactivated and printed flawless. Always used the same GCODE (reuploaded every time).
[octoprint (7).log](https://github.com/paukstelis/Octoprint-Cancelobject/files/2108448/octoprint.7.log)

paukstelis, 2018-06-16:
I've pushed a change to master that should solve the issue of the object regex line being sent to the printer. I can't say with certainty if this will solve your issue, but I suspect it will.

CNCKitchen, 2018-06-17:
Job printed flawlessly with the latest release! Thank you!

closed 2018-06-17 by CNCKitchen