paukstelis/Octoprint-Cancelobject #6: [Enhancement] Add clustering algorithm to detect different parts

JackToaster, 2018-06-22:
Currently, this plugin only works when the gcode is annotated by the slicer to indicate different parts.
I propose using a clustering algorithm to detect different parts and automatically annotate the gcode before printing.
[Here](http://scikit-learn.org/stable/modules/clustering.html#dbscan) is an example of such a clustering algorithm.
This would require parsing the gcode to acquire a list of 4d points (X,Y,Z,time/index) then running a clustering algorithm to split the points into parts, then annotating the gcode.
I'm going to start trying to split gcode files into points as soon as I get a chance, then I'll share my results.

paukstelis, 2018-06-23:
Sounds interesting. I would think you would just want to cluster extrusion moves to simplify things a bit. The first big hurdle I see is that all Slicers have different ways of dealing with extrusion distances which could make things complicated. Looking forward to seeing how it works out.

TTN-, 2018-06-29:
Extrusion distances are either relative or absolute.

I think you could consider a string of moves to be part of a cluster until negative movement is made on the E axis? It could get complicated with settings like retract on layer change, and retracts required that are part of the same object.

paukstelis, 2020-08-25:
Closing as I am not going to look into this.

closed 2020-08-25 by paukstelis