paukstelis/Octoprint-Cancelobject #8: Slic3r support

supermerill, 2018-07-02:
Hi
I was wondering if I can add a pull request to slic3r for the support of your project.
Should i add "; process XXX" or create a more meaningful text, like "; print part XXX". ?

If i modify slic3r to able to add more comments, i have to add (for each layer) a "; print part XXX" at the beginning of each  part print  and a "; end layer XXX" at the end of the layer?

Thanks.

paukstelis, 2018-07-02:
You can add whatever you like since the plugin uses definable regex. My suggestion would be either: `; object XXXX` or `; mesh xxxx`.

There is no need to add the `; end layer`. It just needs to have a comment at the beginning of each block of gcode associated with the object/mesh.

supermerill, 2018-07-03:
My current output for a layer:
`

G1 F8640
G1 X6.333 Y48.685 E-0.27776 ; wipe and retract
G1 E-0.52224 F2100.00000 ; retract
G1 Z0.800 F10800.000 ; lift Z
;AFTER_LAYER_CHANGE
;0.35
M104 S210 ; set temperature
G1 X5.011 Y52.283 ; move to first skirt point
G1 Z0.350 ; restore layer Z
G1 E0.80000 F2100.00000 ; unretract
G1 F900
G1 X4.357 Y51.830 E0.02310 ; skirt
G1 X3.909 Y51.176 E0.02300 ; skirt
G1 X3.723 Y50.402 E0.02310 ; skirt
G1 X3.316 Y5.803 E1.29440 ; skirt
G1 X3.382 Y5.293 E0.01492 ; skirt
G1 X3.566 Y4.813 E0.01492 ; skirt
G1 X3.856 Y4.389 E0.01492 ; skirt
G1 X4.238 Y4.044 E0.01492 ; skirt
G1 X4.645 Y3.816 E0.01355 ; skirt
G1 X5.095 Y3.679 E0.01366 ; skirt
G1 X6.358 Y3.640 E0.03667 ; skirt
G1 X83.905 Y4.061 E2.25059 ; skirt
G1 X84.477 Y4.172 E0.01692 ; skirt
G1 X85.000 Y4.431 E0.01692 ; skirt
G1 X85.433 Y4.816 E0.01682 ; skirt
G1 X85.752 Y5.304 E0.01692 ; skirt
G1 X85.960 Y6.098 E0.02383 ; skirt
G1 X85.961 Y8.377 E0.06614 ; skirt
G1 X85.860 Y8.914 E0.01587 ; skirt
G1 X85.627 Y9.413 E0.01597 ; skirt
G1 X85.277 Y9.838 E0.01597 ; skirt
G1 X84.832 Y10.162 E0.01597 ; skirt
G1 X7.791 Y52.216 E2.54731 ; skirt
G1 X7.036 Y52.458 E0.02300 ; skirt
G1 X5.781 Y52.472 E0.03644 ; skirt
G1 X5.069 Y52.297 E0.02125 ; skirt
; printing object 0 copy 0
G1 F8640
G1 X4.357 Y51.830 E-0.19679 ; wipe and retract
G1 X3.909 Y51.176 E-0.18296 ; wipe and retract
G1 X3.723 Y50.402 E-0.18379 ; wipe and retract
G1 X3.715 Y49.551 E-0.19646 ; wipe and retract
G1 E-0.04000 F2100.00000 ; retract
G1 Z0.950 F10800.000 ; lift Z
G1 X83.151 Y8.021 ; move to first perimeter point
G1 Z0.350 ; restore layer Z
G1 E0.80000 F2100.00000 ; unretract
M204 S800 ; adjust acceleration
G1 F900
G1 X83.151 Y6.471 E0.04846 ; perimeter
G1 X83.551 Y6.471 E0.01251 ; perimeter
G1 X83.551 Y8.021 E0.04846 ; perimeter
G1 X83.211 Y8.021 E0.01063 ; perimeter
M204 S1000 ; adjust acceleration
G1 X83.185 Y7.622 F10800.000 ; move inwards before travel
; stop printing object 0 copy 0
; printing object 0 copy 1
G1 F8640
G1 X83.151 Y6.471 E-0.35817 ; wipe and retract
G1 X83.551 Y6.471 E-0.09236 ; wipe and retract
G1 X83.551 Y7.811 E-0.30947 ; wipe and retract
G1 E-0.04000 F2100.00000 ; retract
G1 Z0.950 F10800.000 ; lift Z
G1 X6.129 Y7.604 ; move to first perimeter point
G1 Z0.350 ; restore layer Z
G1 E0.80000 F2100.00000 ; unretract
M204 S800 ; adjust acceleration
G1 F900
G1 X5.729 Y7.604 E0.01251 ; perimeter
G1 X5.729 Y6.054 E0.04846 ; perimeter
G1 X6.129 Y6.054 E0.01251 ; perimeter
G1 X6.129 Y7.544 E0.04659 ; perimeter
M204 S1000 ; adjust acceleration
G1 X5.767 Y7.434 F10800.000 ; move inwards before travel
; stop printing object 0 copy 1
; printing object 1 copy 0
G1 F8640
G1 X5.729 Y7.604 E-0.09339 ; wipe and retract
G1 X5.729 Y6.054 E-0.35790 ; wipe and retract
G1 X6.129 Y6.054 E-0.09236 ; wipe and retract
G1 X6.129 Y6.991 E-0.21634 ; wipe and retract
G1 E-0.04000 F2100.00000 ; retract
G1 Z0.950 F10800.000 ; lift Z
G1 X6.133 Y48.512 ; move to first perimeter point
G1 Z0.350 ; restore layer Z
G1 E0.80000 F2100.00000 ; unretract
M204 S800 ; adjust acceleration
G1 F900
G1 X6.533 Y48.512 E0.01251 ; perimeter
G1 X6.533 Y50.062 E0.04846 ; perimeter
G1 X6.133 Y50.062 E0.01251 ; perimeter
G1 X6.133 Y48.572 E0.04659 ; perimeter
M204 S1000 ; adjust acceleration
G1 X6.495 Y48.682 F10800.000 ; move inwards before travel
; stop printing object 1 copy 0
;BEFORE_LAYER_CHANGE
G92 E0.0
;0.5

`

paukstelis, 2018-07-03:
Relative extrusion solves the issues.
The only issue I see is that the wipe and retracts associated with object 0 copy 1 are occurring in the object 1 copy 0 block? Not sure how much it matters.

All you would have to change in the settings of the plugin for this would be:
[image]
or if you didn't want `object` to be in the list:
[image]

Have you made these modifications for slic3r or slic3r prusa edition?

supermerill, 2018-07-03:
> Have you made these modifications for slic3r or slic3r prusa edition?

slic3r prusa edition because It's in c++. I avoid perl if possible.

paukstelis, 2018-07-03:
Cool. There is an existing feature request here you can reference with any pull request you make to them:
https://github.com/prusa3d/Slic3r/issues/972

paukstelis, 2018-07-03:
Worked flawlessly for me. 
1. Make sure you use relative extrusion. I have a version that will work with absolute, but haven't integrated it into master branch yet.
2. Use some Z-hop. Not required, but it is helpfully for resetting Z-height. If you don't use Z-hop, make sure to add `G1 Z` to the allowed gcode section in settings
3. I added `; printing object ENDGCODE` to the start of the ending gcode block in custom gcode.

The 4 objects:
[image]

The plugin settings:
[image]
I let two layers print for each object, then cancelled the first and third objects:
[image]

Results look good. I did cancel it before it finished, but you see that it works as expected.
[image]

paukstelis, 2018-07-03:
It would be awesome to pass object (file) names instead of just their index if at all possible.

supermerill, 2018-07-03:
I didn't find the name string. I was lucky enough that their index is preserved...
I will try to see if i can push it from the gui to the gcode writer.

paukstelis, 2018-07-03:
Nice!

supermerill, 2018-07-05:
Now merged into slic3r original.

Tinchus2009, 2018-07-09:
mmmm I see a potential issue, only "potential": will this pluing work if we have a set of object, and each object is printed with different layer heights?

supermerill, 2018-07-09:
> and each object is printed with different layer heights?

how can you do that?

paukstelis, 2018-07-09:
I have done this (with Simplify3D processes) and it makes no difference. The plugin just skips regions of the gcode. The Z-height will get reset before each object.

Tinchus2009, 2018-07-09:
easy: in each object you can assign it own properties, one of them is the custom layer heights. Double click the object > then you can choose to do it in 2 different ways: adaptative layer or by the layer height table. So in 1 print with multiple objects in the bed, you can have different object printing with different heights.

Also Can someone clarify something about this plugin?: "Gcode (comma delimited) can be injected before or after the server reaches a cancelled object. This may be need for retractions or resetting extrusion distance in some cases." Wich cases are we talking about here? examples?

paukstelis, 2018-07-09:
I wish I could be more specific, but the gcode injection is really just there for special need cases. I have yet to come across a case with a slicer that I have used (S3D, Slic3r, Cura) where I have needed to use it. I have written a version of the plugin that works with absolute extrusion that uses these to set the extrusion distance after a skipped object.

supermerill, 2018-07-09:
@Tinchus2009 
Seems okay
```

M900 K30; Filament gcode
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
;BEFORE_LAYER_CHANGE
G92 E0.0
;0.2

G1 E-0.80000 F2100.00000 ; retract
G1 Z0.600 F10800.000 ; lift Z
;AFTER_LAYER_CHANGE
;0.2
; printing object TestCylinder.stl id:0 copy 0
G1 X121.049 Y106.277 ; move to first perimeter point
G1 Z0.200 ; restore layer Z
G1 E0.80000 F2100.00000 ; unretract
M204 S1000 ; adjust acceleration
G1 F900
G1 X121.095 Y106.239 E0.00224 ; perimeter
G1 X121.161 Y106.220 E0.00262 ; perimeter
G1 X120.651 Y106.256 F10800.000 ; move to first perimeter point
G1 F900
G1 X120.698 Y106.123 E0.00531 ; perimeter
G1 X120.829 Y105.963 E0.00780 ; perimeter
G1 X121.012 Y105.865 E0.00780 ; perimeter
G1 X120.261 Y106.284 F10800.000 ; move to first perimeter point
G1 F900
G1 X120.295 Y106.109 E0.00671 ; perimeter
G1 X120.364 Y105.944 E0.00671 ; perimeter
G1 X120.463 Y105.796 E0.00671 ; perimeter
G1 X119.884 Y106.344 F10800.000 ; move to first perimeter point
G1 F900
G1 X119.884 Y106.247 E0.00365 ; perimeter
G1 X119.933 Y105.999 E0.00951 ; perimeter
G1 X120.030 Y105.766 E0.00951 ; perimeter
G1 X119.966 Y106.292 F10800.000 ; move inwards before travel
; stop printing object TestCylinder.stl id:0 copy 0
;BEFORE_LAYER_CHANGE
G92 E0.0
;0.3

G1 F8640
G1 X119.884 Y106.247 E-0.03628 ; wipe and retract
G1 E-0.04000 F2100.00000 ; retract
G1 Z0.800 F10800.000 ; lift Z
;AFTER_LAYER_CHANGE
;0.3
; printing object TestCylinder.stl id:1 copy 0
G1 X127.583 Y104.677 ; move to first perimeter point
G1 Z0.300 ; restore layer Z
G1 E0.80000 F2100.00000 ; unretract
G1 F900
G1 X127.569 Y104.537 E0.00747 ; perimeter
G1 X127.610 Y104.403 E0.00747 ; perimeter
G1 X127.699 Y104.294 E0.00747 ; perimeter
G1 X127.295 Y104.910 F10800.000 ; move to first perimeter point
G1 F900
G1 X127.241 Y104.781 E0.00748 ; perimeter
G1 X127.214 Y104.643 E0.00748 ; perimeter
G1 X127.214 Y104.502 E0.00748 ; perimeter
G1 X127.033 Y105.158 F10800.000 ; move to first perimeter point
G1 F900
G1 X126.980 Y105.079 E0.00506 ; perimeter
G1 X126.899 Y104.884 E0.01121 ; perimeter
G1 X126.858 Y104.678 E0.01121 ; perimeter
G1 X126.770 Y105.405 F10800.000 ; move to first perimeter point
[... 210 lines cut ...]
G1 Z1.500 F10800.000 ; lift Z
G1 X128.467 Y105.848 ; move to first infill point
G1 Z0.900 ; restore layer Z
G1 E0.80000 F2100.00000 ; unretract
M204 S2000 ; adjust acceleration
G1 F2400
G1 X128.198 Y105.932 E0.00353 ; infill
G1 X128.094 Y105.943 E0.00131 ; infill
G1 X126.651 Y104.033 E0.00353 ; infill
M204 S1000 ; adjust acceleration
; stop printing object TestCylinder.stl id:1 copy 0
G1 F8640;_WIPE
G1 X126.567 Y104.302 E-0.06506 ; wipe and retract
G1 E-0.04000 F2100.00000 ; retract
G1 Z1.500 F10800.000 ; lift Z
M107 ; disable fan
; Filament-specific end gcode
G4 ; wait
M104 S0 ; turn off temperature
M140 S0 ; turn off heatbed
M107 ; turn off fan
G1 Z31.5 ; Move print head up
G1 X0 Y200; home X axis
M84 ; disable motors
```

gdombiak, 2018-11-08:
Do you guys know when Slic3r PE might merge and add support for this cool plugin?

supermerill, 2018-11-09:
@gdombiak 

currently you can have access to this with the mainline slic3r (https://dl.slic3r.org/dev/)
or my slic3rPE fork (https://github.com/supermerill/Slic3r/releases)

> Do you guys know when Slic3r PE might merge

The prusa guys are busy doing their migration from perl to c++ and integrating the prusacontrol interface, manual supports also. Maybe in a few months they can look at something else.

gdombiak, 2018-11-09:
Nice to see that you are keeping the fork up to date. I'm on a Mac so will give it a try once I have windows on VirtualBox.

Thanks for your work. Once this gets merged into the official Slic3r PE a lot more people will use this great plugin. Hope that happens sooner than later.

supermerill, 2019-02-06:
Just for reference, it's now also added into the last alpha of slic3rPE_1.42.0

Tinchus2009, 2019-02-06:
sorry. What has been exactly added to the PE?

supermerill, 2019-02-06:
Now you have an option dedicated to enable/disable the needed comments in the Output options.

closed 2020-08-25 by paukstelis