paukstelis/Octoprint-Cancelobject #9: Plugin tab blank after installation, cryptography error

nicksears, 2018-07-05:
I have a relatively new octopi install, but have been getting a cryptography error for a lot of plugins:

IT says installed successfully "Successfully installed OctoPrint-Cancelobject-0.1.1" but then says: 

"pyopenssl 18.0.0 has requirement cryptography>=2.2.1, but you'll have cryptography 2.1.4 which is incompatible."

and then:

"Done!
Close
Plugin management log
Install Unknown: 
A restart is needed for the changes to take effect.

Restart now"

After restart the plugin shows up but the page is blank when the tab is selected. Can provide the full install log, but no other errors...

paukstelis, 2018-07-05:
The tab is blank until you start printing. The objects will be populated when the print starts.

Psyc0Flyer, 2018-07-05:
Have you done sudo apt-get update and sudo apt-get upgrade? I know this has fixed many issues I have had.

nicksears, 2018-07-06:
Tried that but now luck. also tried updating to maintenance RC channel, but not luck.

After trying to update 1.3.9rc1 to rc2 my pi would boot but no octoprint server (no python?)

so I just reinstalled everything and it installed fine, working fine now...

nicksears, 2018-07-06:
scratch that. It installs fine (doesn't say "Unknown" anymore), says "Cancelobject"

Slic3r is correctly inserting lines like "; printing object E3D40mmFanMount (BEST).stl id:0 copy 0"

and I've changed the regular expression from  "; process (.*)"  to  "; printing object (.*)"

but it still shows a blank page when the tab is selected...

paukstelis, 2018-07-06:
object regex needs to be `; printing object (.*)`
You are missing the asterix.

nicksears, 2018-07-06:
yes, sorry, it's github's formatting. I did include the asterisk.

paukstelis, 2018-07-06:
Re-download your gcode after you have uploaded and take a look to see if it has changed the object regex lines to `#Object xxxxxxx`

You are loading the gcode file and/or starting a print before you look at the Cancel Object tab, right?

nicksears, 2018-07-06:
this is after loading the file, and I just started a print, too with no change.

When I redownload the gcode I don't see "#object" (just the original "; printing object E3D40mmFanMount (BEST).stl id:0 copy 0".

I also don't see the active object in the navbar, and that used to be there even when it hadn't installed right.

Now I need to wait for the print to finish, actually need this part...im sure it made me reboot octoprint, but maybe ill try and reboot the pi as well?

Will also try update/upgrade, but it'll be a bit. So excited to get this working, thanks for all the help and quick work!

paukstelis, 2018-07-06:
Please provide a log file (octoprint.log) that includes information after/during upload a file and when the gcode file is loaded.

paukstelis, 2018-07-06:
Also, make sure you re-upload the file after you restart. The tags get changed right after the upload.

nicksears, 2018-07-06:
It works it works! Can't explain it, I just had to reinstall.

In case anyone else runs into the same issue, it was definitely doing something with the file. When I refreshed it was reading the filesize as "-" and downloading it had 0kb...

Very excited to try this out. [This guy ](https://www.youtube.com/watch?v=ANfOr2F79LQ)isn't exaggerating when he says how impactful this feature is.

closed 2020-08-25 by paukstelis