paukstelis/Octoprint-Cancelobject #11: Sticky Processes in Cancel Objects Tab

phcoetzee, 2018-07-14:
Thank you paukstelis for a great plugin.

I have no issues with the plugin, only a suggestion, example:
When I've uploaded a gcode to octoprint to print but decided not to print it and delete it straight away (before clicking on print), the different processes of that gcode file stays in the Cancel Objects Tab and doesn't go away when you delete the gcode file. Even a reboot or F5 doesn't remove it. Only when I load another gcode to be printed that previous processes go away. It will be nice if the processes go away same time as when delete gcode before print. It is not an issue for me, just a suggestion to you (I know you are a very busy guy).

Thank you

paukstelis, 2018-07-15:
I will look into it. Should be a straightforward fix once I figure out what the event is for that.

paukstelis, 2018-07-23:
This is fixed in the latest version. Please reinstall and confirm if possible.

phcoetzee, 2018-07-23:
Thank you @paukstelis , I can confirm that the issue has been resolved. Processes in the Cancel Objects Tab now go away if you delete a gcode prior to printing.
Thank you for your time and patience.

closed 2018-07-23 by phcoetzee