paukstelis/Octoprint-Cancelobject #13: Expose objects via API

gdombiak, 2018-10-03:
Hi,

I'm the developer of [OctoPod (free open source iPhone app for OctoPrint)](https://github.com/gdombiak/OctoPod), and I would like to expose your plugin in the app. I was looking at the source code and I see an API call for cancelling an object. Maybe I missed it, but I do not think I saw an API for retrieving the list of objects.

This issue is for adding this new API. I will then add your plugin to OctoPod. :)

Great job
Gaston

paukstelis, 2018-10-03:
It think being able to retrieve an object list via API would be pretty straightforward, though I'll have to look again at how things were organized. Is there anything else that would be required for OctoPod?

gdombiak, 2018-10-03:
Thanks for the quick response and great to hear that can be easily done. :)

I was thinking in general lines that OctoPod will:

1. Get list of objects that can be cancelled (previously cancelled are gone from the list or have some status indicating that they were cancelled) - requested by this issue
1. Be able to request to cancel an object  - existing API does this
1. Get confirmation that cancellation was successfully processed (usually plugins provide this info via websockets) - Not sure if this is implemented or if needed in case cancellations never fail

If you would like to test the plugin with OctoPod (once updated) before releasing the new version of the plugin let me know and we can coordinate something.

gdombiak, 2018-11-05:
Hi Paul,

I'm working on the 2.0 version of OctoPod and would love to add support for your plugin in this release. Not sure if our timing will work out or not. I'm happy to help test this or provide more feedback.

Thanks,
Gaston

paukstelis, 2018-11-05:
Sorry, Gaston. I've just been busy with work and other big projects. I'll see if I can carve out some time in the next couple of weeks. The biggest problem is remembering how everything works! I'm getting old...

paukstelis, 2018-11-05:
I'm not currently in a position to test this, but you could can checkout the octopod branch. I just added a couple of lines that I think should provide a json array of the current object list using the `objlist` api command.

gdombiak, 2018-11-05:
That's great. I will test the _octopod_ branch with your change and let you know how it goes. 

Are there instructions on how to build the plugin? I never created an OctoPrint plugin from source. I have a local OctoPrint running from PyCharm or an instance running from OctoPi.

Thanks,
Gaston

gdombiak, 2018-11-08:
Seems like the problem is that **flask.jsonify** expect a dictionary instead of an array. Changing line 146 to this works fine:

```python
			return flask.make_response(flask.jsonify(dict(list=self.object_list)))
```

gdombiak, 2018-11-08:
Great news! OctoPod now supports this plugin. Let me know when you are planning to push the changes we discussed in this ticket. I can help spread the word when releasing the new version of OctoPod.

[image]

paukstelis, 2018-11-08:
changes pushed to master

gdombiak, 2018-11-08:
That's great! When do you think you will make the next release?

gdombiak, 2018-11-11:
Heads up, my OctoPrint found the new release of your plugin but it got a 404 installing the new version. Here is the error message I got:

> Now updating Cancel object to 0.2.1
+++++++++++++++++++++++++++++++++++
/home/pi/oprint/bin/python2 -m pip install https://github.com/you/OctoPrint-Cancelobject/archive/0.2.1.zip --no-cache-dir
HTTP error 404 while getting https://github.com/you/OctoPrint-Cancelobject/archive/0.2.1.zip
Could not install requirement https://github.com/you/OctoPrint-Cancelobject/archive/0.2.1.zip because of error 404 Client Error: Not Found for url: https://github.com/you/OctoPrint-Cancelobject/archive/0.2.1.zip
Could not install requirement https://github.com/you/OctoPrint-Cancelobject/archive/0.2.1.zip because of HTTP error 404 Client Error: Not Found for url: https://github.com/you/OctoPrint-Cancelobject/archive/0.2.1.zip for URL https://github.com/you/OctoPrint-Cancelobject/archive/0.2.1.zip

paukstelis, 2018-11-11:
Yeah, I just messed up. It is now fixed so it won't happen with future releases. I'll edit the releases notes.

closed 2019-08-31 by paukstelis