paukstelis/Octoprint-Cancelobject #15: How to use with Cura ?

Nuttavoot, 2018-10-12:
Can someone explain how to use this plug-in with Cura, I could not understand how to do it ?

paukstelis, 2018-10-12:
To use it most effectively with Cura you'll need to download or compile the
current master branches of Cura and CuraEngine from github.

giloserCGM, 2018-10-12:
so it's not in the current 3.5 or 3.4.1 version?

paukstelis, 2018-10-12:
I don't believe so. They merged my pull request only a few days before
announcing 3.5. I don't know if they keep snapshot builds somewhere of the
most recent master branches. If someone knows, please point me to them.
Building CuraEngine would be a pain for most people.

Nuttavoot, 2018-10-13:
I don't know how to compile Cura, I have to wait or use other slicers. Can I use with SLIC3R Prusa Edition ?

giloser, 2018-10-17:
Hi again,
I have not try to compile Cura as well, it's really in the CuraEngine, not possible to put it on a plugin or something like that?

anyway we'll wait for it to be implemented in one version of Cura if we have to.

thanks a lot!

paukstelis, 2018-10-17:
I'm afraid not. All gcode from Cura comes from CuraEngine.
The good news it that it should certainly be in the next full release.

Skullmonkey, 2018-11-04:
Hello,
Is there a way to compile the code? It seems that the URL comes back to a 404 and there are no close matches in the branches. Really looking forward to be able to use this plugin and save some filament.

Thanks

paukstelis, 2018-11-05:
It seems that the most recent beta was supposed to have all the changes in it (I was listed as a contributor for my commits) but it looks like even that version doesn't have the CuraEngine modifications in it. At least when I tested it was not outputting the proper comments.

Nuttavoot, 2018-11-05:
We can use this plugin with Cura 3.6 beta, is it correct ?

paukstelis, 2018-11-05:
Yes. If you use the latest version of the plugin it should work out of the box with Cura 3.6. The only thing that I know is an issue involves the way Cura does supports. They aren't linked to a particular object, so cancelling an object may not cancel its supports.

eliliam, 2019-06-07:
Using the latest Cura 4.1 (Parts in gcode separated by ";MESH:<name_of_stl + some amount of (0)'s added"), I am still not able to get any of my objects recognized inside of the plugin tab. Is there some custom regex I need to use in the settings page of this plugin? If so, can someone point me in the right direction?

paukstelis, 2019-06-07:
How are you uploading to your octoprint instance?

eliliam, 2019-06-07:
Export from cura -> open octopi through browser -> click on upload(to pi, not to internal SD)

eliliam, 2019-06-07:
I am unsure if this has anything to do with it, but there are some console errors in regards to this plugin.
https://pasteboard.co/IimIPxI.png

paukstelis, 2019-06-07:
Download the file from Octoprint and check to see if it contains the `@` commands that replace the ;MESH comments.

eliliam, 2019-06-07:
Yes it does seem to have replaced those ;MESH lines with @Object <name_of_mesh>

paukstelis, 2019-06-07:
The image link is not working for me. Can you paste the errors here? I

eliliam, 2019-06-07:
`Could not instantiate the following view models due to unresolvable dependencies: packed_core.js:13421:25
    <anonymous> https://octopi.local/static/webassets/packed_core.js?0cc83181:13421
    mightThrow https://octopi.local/static/webassets/packed_libs.js?ea9fbc9a:3535
    process https://octopi.local/static/webassets/packed_libs.js?ea9fbc9a:3603
`

`cancelobjectViewModel (missing: gcodeViewModel ) packed_core.js:13423:29
    <anonymous> https://octopi.local/static/webassets/packed_core.js?0cc83181:13423
    Pn https://octopi.local/static/webassets/packed_libs.js?ea9fbc9a:11208
    ur https://octopi.local/static/webassets/packed_libs.js?ea9fbc9a:11229
    <anonymous> https://octopi.local/static/webassets/packed_core.js?0cc83181:13422
    mightThrow https://octopi.local/static/webassets/packed_libs.js?ea9fbc9a:3535
    process https://octopi.local/static/webassets/packed_libs.js?ea9fbc9a:3603
`

paukstelis, 2019-06-07:
Did you remove the gcodeviewer?

eliliam, 2019-06-07:
I just re-enabled gcodeviewer to see if that fixed anything, but the plugin doesnt seem to work with or without gcodeviewer enabled.

paukstelis, 2019-06-07:
Are you getting anything different in console now?

eliliam, 2019-06-08:
Nope, no change, was there something you wanted me to do that would have changed the console?

paukstelis, 2019-06-08:
That initial error appears to be from gcodeviewer not being present. I would expect it to go away once that was fixed.

Please provide any console errors from your browser and your octoprint.log file after a restart. That might provide some clues.

If you have a highly custom installation its harder to debug. The plugin works with the basic octoprint installation quite readily.

closed 2020-08-25 by paukstelis