paukstelis/Octoprint-Cancelobject #16: Can not update to new version.. 0.2.1

Stady234, 2018-11-09:
Updating...
Updating, please wait.
+++++++++++++++++++++++++++++++++++
Now updating Cancel object to 0.2.1
+++++++++++++++++++++++++++++++++++
/home/pi/oprint/bin/python2 -m pip install https://github.com/you/OctoPrint-Cancelobject/archive/0.2.1.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/you/OctoPrint-Cancelobject/archive/0.2.1.zip
HTTP error 404 while getting https://github.com/you/OctoPrint-Cancelobject/archive/0.2.1.zip
Could not install requirement https://github.com/you/OctoPrint-Cancelobject/archive/0.2.1.zip because of error 404 Client Error: Not Found for url: https://github.com/you/OctoPrint-Cancelobject/archive/0.2.1.zip
Could not install requirement https://github.com/you/OctoPrint-Cancelobject/archive/0.2.1.zip because of HTTP error 404 Client Error: Not Found for url: https://github.com/you/OctoPrint-Cancelobject/archive/0.2.1.zip for URL https://github.com/you/OctoPrint-Cancelobject/archive/0.2.1.zip
The update did not finish successfully. Please consult <code>octoprint.log</code> and <code>plugin_softwareupdate_console.log</code> for details.

paukstelis, 2018-11-09:
It was a simple fix, but unfortunately, you'll need to uninstall/reinstall the plugin. Future versions won't have this issue.

Stady234, 2018-11-10:
Awesome, got everything reinstalled!  Thanks!

closed 2018-11-10 by Stady234