paukstelis/Octoprint-Cancelobject #19: No visualization

Popeyef5, 2018-12-10:
I installed the plugin but am unable to see the objects displayed. I am generating the GCODEs through a slic3r PE fork which supports the plugin. This way, it is difficult to know which object to cancel when one is faulty since I only have information of which copy is being printed at the time. Is this normal? Is there a way to display the copies on the bed?

paukstelis, 2018-12-10:
This is normal. However, you can at least see when it is printing on a failed object and know that is the one you should cancel. I had hoped to work on a version that would display the name of object at its center of mass in the gcode viewer (and allow cancelling from there), but I simply haven't had time to figure out how to do that yet. It requires pre-parsing of the gocde which is something that is not impossible but is more involved.

Popeyef5, 2018-12-10:
Oh, I understand. I know little to nothing, but how about displaying a rectangle and simply displaying tags at the average of the X-Y coordinates of the different objects and that updates as the GCODE goes on?

To make myself clear:

Would it be possible to have a 2D rectangle with the dimensions of the bed and tags displayed on top of it? Each tag would correspond to a copy of an object. The position of each tag would be the median of the XY coordinates of the GCODE lines corresponding to the tag's copy and already printed (no pre-processing needed).

I would be happy to ellaborate further if you want. Sorry if it's a bit messy, I don't have much time now but later I will.

paukstelis, 2018-12-10:
Yes, I understand what you are saying. It would be possible to do it that way, though there would be a bit more overhead on the plugin to keep updating the object's position. This is probably most easily integrated into the existing gcode viewer than trying to make something from scratch, though that viewer has some issues of its own (large files).

paukstelis, 2019-02-12:
As requested, I have a basic implementation that uses the existing gcode viewer to cancel objects. I'm looking for feedback to see if/how it works for others in real scenarios. You can install it from the postiontrack branch.

closed 2020-08-25 by paukstelis