paukstelis/Octoprint-Cancelobject #20: MakerGear M3-ID not connecting

Deduin42, 2019-01-02:
With the plugin loaded and active, Octoprint is not able to connect successfully to my printer - state hangs at "connecting".

If plugin is disabled or uninstalled, connection works normally.

MakerGear M3-ID, connecting via local IP address
Octoprint v 1.3.6

paukstelis, 2019-01-02:
1.Update your octoprint installation. 1.3.6 is very old.

2. What is the plugin version? There really shouldn't be any reason this plugin would do anything with connecting to the printer. If you haven't, uninstall and reinstall the plugin from the plugin repository.

Deduin42, 2019-01-02:
I've uninstalled and reinstalled a couple times to make sure, but for whatever reason with it installed and active, the printer won't connect.

I'll have to see if I can update Octoprint.  right now it's running the latest MakerGear modified version.

paukstelis, 2019-01-02:
Provide your octoprint.log file. That should provide some clues.

Deduin42, 2019-01-02:
Here's a copy.  Let me know if the attachment doesn't work. Thank you for your quick responses!

[octoprint.log](https://github.com/paukstelis/Octoprint-Cancelobject/files/2721440/octoprint.log)

paukstelis, 2019-01-02:
The only error I see is:
```
2019-01-02 16:57:55,130 - octoprint.util.comm - ERROR - Error while processing hook cancelobject for phase queuing and command M110 N0:
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.6-py2.7.egg/octoprint/util/comm.py", line 2419, in _process_command_phase
    hook_results = hook(self, phase, command, command_type, gcode, subcode=subcode)
```
This is probably because it is an old version of octoprint. I'm not sure why that error would prevent the printer from connecting.

Deduin42, 2019-01-02:
I appreciate you looking into it.  For now, I think I'll close the issue ticket, but if I can update to the newer Octoprint and it works, I'll make sure to send you the results.  I'm still pretty new to using Octoprint, so I want to be cautious to not mess it up. :)

closed 2019-01-02 by Deduin42