paukstelis/Octoprint-Cancelobject #22: fix pause at object omission broken

SirBastol, 2019-01-14:
when printing and cancelling object it stops printing for the same amount of time it would take to ahve continued to print it.. the head stays put and just dribbles filament(oozing)
 
one would assume that when you cancel an object it would not only skip that section of code but  not process it as if its live with timings etc.. 

it should skip over that area almost instantly 

this i feel is a bug.. 

it would be nice to add comparability to gcode viewer and their cancel area options plugin  to add a hover to see object name and right click to cancel it.. or a button to let you cancel objects from there.. ( instead of having several places to go . you could even add your table list selector to the bottomof that page if its installed ?

note in gcode viewer editor it shows the layers printing as normal but the printer doesn't do anything.. and removing or disabling the viewer doesn't change the end result of head not moving or doing anything while that section of code is processed.

paukstelis, 2019-01-14:
There is no reason it should do that, and it certainly doesn't do that in any other cases I have heard of. Please provide your gcode and a list of other plugins you have installed. It sounds to me likely to be a compatibility issue.

SirBastol, 2019-01-15:
i have removed all plugins other than the default. the function is the same
i only have installed octolapse and print details logging ( pritnerstats)
andgcode viewer as a base..

Paul Cole
http://ca.linkedin.com/in/paulcolevfx
--=-=-=-=-=-=--
http://db.tt/LXIuAye0
^^click me for free cloud storage^^

“Defeat is a state of mind. No one is ever defeated until defeat has been
accepted as reality. "
- Bruce Lee
"http://tools.ip2location.com/ib2"  <http://tools.ip2location.com/ib2>

paukstelis, 2019-01-15:
You'll need a to provide a lot more information because as far as I know you are the only one with this issue.

What version of octoprint? What type of controller are you using? What kind of printer is it?
At a minimum provide your octoprint.log file (after you have attempted to cancel an object), and the gcode file that you are using. Also, provide your settings for the plugins.

SirBastol, 2019-01-18:
It does the same thing with nothing installed except the latest octoprint
and your plugin

It does skip the code but it pauses for the duration it woulda took on the
cancel

-- 

Paul Cole
http://ca.linkedin.com/in/paulcolevfx
--=-=-=-=-=-=--
http://db.tt/LXIuAye0
^^click me for free cloud storage^^

“Defeat is a state of mind. No one is ever defeated until defeat has been
accepted as reality. "
- Bruce Lee
"http://tools.ip2location.com/ib2"  <http://tools.ip2location.com/ib2>

paukstelis, 2019-01-18:
Again, please provide the gcode you are using and the relevant log files. I can't help figure out what's going on without more information.

As a first blind guess, are you using relative extrusion in whatever slicer you are using?

SirBastol, 2019-01-18:
Cura latest version. To octoprint. Default settings  most time. 210/55. And 60mm/s 
.2mm layer

And haven’t had logs since I wiped it beecause you said it was a plug-in

Sent from my iPhone please excuse typo mess or voice to text errors.

mkgin, 2019-02-13:
I noticed this too. Next time I need to cancel an object, I'll have a look at the logs.

paukstelis, 2019-02-13:
The only reasoning that I have for why you would have a long pause would be it is a very large object, or if you are cancelling a sequential print where it is effectively cancelling many layers at once. Its also possible that if there is anything defined in the "Allowed GCODE" blocks that it is getting executed as those commands are allowed to come out of the queue. Logs would be useful. I'm going to add the beginnings of a debug mode in the next release so it will provide some more information.

mkgin, 2019-02-17:
Hi,

When this happened, I remember that I was printing objects where I had set different infill and perhaps perimeters , for the objects in the being printed.  I may add a small test object when I I use the printer next ( next week some)  and see if I can reproduce it by cancelling an object.  I print all the objects at the same time. 

I did use the plugin to cancel a print  later on where... no real delay, but the where where extra moves to  near the  canceled object, likely because I had avoid crossing perimeters set... So, it depends where slicer decides that it begins the new object.

Haven't really looked in detail into how Octoprint plugins work or how CancelObjects is implemented, but understand enough about what the plugin would need to do. I guess that it operates like a middleman on the stream of data  and running a regex... It guess it wouldn't operate on the gcode.file unless there 

By large object, do  you mean many lines of GCODE for that object (or  for that object in that layer)

I suppose it is not difficult to have lots of lines of  GCODE per object even in one layer, especially if using high detail,  circles or complex infill patterns.  

There are many places external to the plugin where there could a bottlenecks. For example IO  to the printer or the flilesystem (which is typically mostly on the whatever SDCARD the user choses).

I am running on Octoprint on a PI connected to a Prusa i3 MK3 via the the rambo board's  USB connection.

After some reading in the prusa forums, I learned that the serial USB connection is slow and not reliable. Also that in some cases Octoprint has problems when a command on the printer takes longer to run than Octoprint expects, and Octoprint thinks there is a communication problem.

Maybe this plugin isn't the best place to for this type of functionality, but it could be worth logging a warning when communication is slow or there is a lot in the buffer to send to the printer. Maybe a pop up notification with a checkbox too  to not be alerted about this if it happens again with the same object or same print job.

paukstelis, 2019-02-17:
The extra moves near the cancelled object are usually wipe moves that the slicer puts after its completed extrusion moves for the object. I don't know how configurable that is in slic3r but @supermerril might be able to comment. Unless your are printing at insanely fast speeds, serial via USB is perfectly fine and totally reliable. The newest version of the plugin uses octoprint `@` commands to tag objects so those never get sent to the printer.

What kind of Pi are you running? I wouldn't be surprised if something that is a bit underpowered (Zero W) had more problems than 3b. I personally run nearly all my printers using octoprint off of various linux boxes (9 printers off of one old core 2 duo) with no communications issues whatsoever.

mkgin, 2019-02-18:
Raspberry Pi 3 Model B... It was on special just before the 3 B+ cam out :-/... Not printing fast.
Wifi and USB serial to the printer.

With the PI, Ethernet, Wifi, serial share a single USB bus (the hub is part of the Ethernet Soc if I remember right), this is a known bottleneck on the PI.  Your Core Duo probably has 4 usb buses.

paukstelis, 2019-02-18:
Only 2 buses on the Core Duo, but I added a third via PCI. It really doesn't take all that much USB bandwidth to stream commands from octoprint. I have run up to 2 YUV webcams which use/reserve a ton and still happily run 3 printers simultaneously all on the same bus.

paukstelis, 2020-08-25:
Closing, no new info on this.

closed 2020-08-25 by paukstelis