paukstelis/Octoprint-Cancelobject #23: Canceling Supports

agrabbs, 2019-01-22:
Not sure if this is a possibility, but it appears the Cancel Object only cancels the object and not the support(s) as well; The supports layer appears as "NONMESH" and may be inseparable object to object.

I canceled an object and it still keeps printing the support.

paukstelis, 2019-01-22:
Yes, that is an issue with Cura, unfortunately. Their slicing engine considers supports a separate object and is blind to what object it is actually supporting. That isn't he case for Simplify3D or Slic3r.

closed 2019-01-22 by agrabbs

paukstelis, 2019-01-22:
I mostly use Cura and Slic3r PE. I used S3D a lot in the past, but I'm pretty much fed up with their lack of updates.

agrabbs, 2019-01-22:
Thanks for all your help. Wish Cura would get on board, I like their software.

paukstelis, 2019-01-22:
Yeah, they are developing for a pretty specific platform, so I understand where they are coming from. They are receptive to changes if people do the work (I provided the commits to get the mesh names output into the gcode). I thought about trying to tackle the support issue but it is so woven into how CuraEngine slices it would require a huge rewrite.

agrabbs, 2019-01-22:
Well, for whatever it's worth I think your work is awesome and I appreciate the work you do.