paukstelis/Octoprint-Cancelobject #26: Cura[3.6.0] order of auto-arranged models

JLIspace, 2019-02-22:
Hey,
I just tried it the first time (after a had to stop a pretty long print because of one small part 😐). And as it turns out it works great! The only thing I had to change inside cura was relative extrusion as someone pointed out already (I guess, haven't tried it otherwise).
My only 'problem' is that I can't really differentiate a lot of (auto arranged) copys of the same stl file. Cura just tags them as "xxx.stl (1)", "xxx.stl (2)". I thought that this would happen in the order of printing, but that's not the case - at least for me.

So is there any easier of spotting the right (already broken part) to cancle, without having to wait until the "red box" will appear just seconds before the printer is going to crash into it again?

Regards! :)

paukstelis, 2019-02-22:
This will be addressed in latest version that will be out (hopefully) next week. It will provide markers in the gcodeviewer that you can use to cancel objects directly. Right now I am just trying to make it compatible with the ExcludeRegion plugin.

JLIspace, 2019-02-22:
Oh, okay. That sounds awesome! 👍

closed 2019-02-22 by JLIspace

paukstelis, 2019-02-27:
New version has been released that allows cancelling from gcodeviewer. Please let me know how it works for you.

JLIspace, 2019-02-28:
That's way better! There is no need to think about the order anymore.
Additionally, you could show the file names in the "Are you sure?" window so you can cancel similar looking (but different) models when you change your mind afterwards.
This would also remove the chance of aborting the wrong model due to tight spacing or wrong selection.

Thanks for the update!

paukstelis, 2019-02-28:
yeah, adding object name in the pop-up should be pretty straightforward. I'll add that to the TODO for the next release.

paukstelis, 2019-03-08:
You can try out the feature in the 0.3.2 branch. Install from octoprint via this url: https://github.com/paukstelis/Octoprint-Cancelobject/archive/0.3.2.zip