paukstelis/Octoprint-Cancelobject #27: Update fail

kb-, 2019-02-25:
OctoPrint is asking in loop to update Cancelobject. It says it's installing v0.2.1 instead of 0.3.

I'm using OctoPrint 1.3.10
And OctoPi 0.15.1, running on Raspberry Pi 3 Model B Plus Rev 1.3

paukstelis, 2019-02-25:
Well, that's annoying. I seem to remember doing this once before but I don't remember how I fixed it....

paukstelis, 2019-02-25:
Can you confirm that it will now update to 0.3.1?

kb-, 2019-02-25:
Not directly, but yes.

It offered me to install 0.3.0 first, did the same thing with 0.2.1 and finally offered me to install 0.3.1. My current version is now 0.3.1.

djokoss22, 2019-03-07:
Update fail for me too when i try to update from OctoPrint.

I found that OctoPrint try to download from https://github.com/you/Octoprint-Cancelobject
but the real URL is https://github.com/paukstelis/Octoprint-Cancelobject

So, i did the update manually by typing this command in the Raspberry Pi:
`/home/pi/oprint/bin/python2 -m pip install https://github.com/paukstelis/Octoprint-Cancelobject/archive/0.3.1.zip --no-cache-dir`

Now, Cancelobject is in 0.3.1 version, ;)

paukstelis, 2019-03-07:
This was a known issue with the initial release. You should get future updates without issue now.

> Update fail for me too when i try to update from OctoPrint.
>
> I found that OctoPrint try to download from https://github.com/you/Octoprint-Cancelobject/archive/0.3.1.zip
> but the real URL is [...]

paukstelis, 2020-07-18:
Closing this. I think I have figured out how to do properly releases now.

closed 2020-07-18 by paukstelis