paukstelis/Octoprint-Cancelobject #28: [Request] Toggle orange circle in gcode viewer

calania, 2019-03-05:
Hello, I really like that you added the orange circle over objects in the gcode viewer but sometimes, I want to hide it, for example, if there is a detail under it I want to see. It would be great could you add something like a "toggle current object marker" button somewhere in the gcode viewer.

paukstelis, 2019-03-05:
That should be easy enough. I'll add it to the list.

calania, 2019-03-05:
That was a fast reply, thank you!

paukstelis, 2019-03-08:
You can try out this new feature in the 0.3.2 branch. You can install in octoprint via url here: https://github.com/paukstelis/Octoprint-Cancelobject/archive/0.3.2.zip

calania, 2019-03-08:
Thanks for implementing it so quickly however I can't get it to work. When I install it through your link it says that it is downloading version 0.3.2 but then, when it's done, it says that it has successfully installed version 0.3.1 and I can't find any way to hide the marker.

paukstelis, 2019-03-08:
That's weird, I was sure I pushed the commit yesterday put apparently it wasn't there. If  you try again it should work now.

calania, 2019-03-08:
Tried it now and it seems to be working fine. Thanks for implementing it and adding it so quickly!

closed 2019-03-08 by calania