paukstelis/Octoprint-Cancelobject #30: How to use with Slicer PE?

guestisp, 2019-03-29:
I'm trying to use this plugin with Slic3r PE, latest beta.

In the GCODE I can see:
```
; stop printing object AIl_In_One_3D_printer_test_micro.stl id:0 copy 0
;M117 Printing layer 4
;LAYER:4
G1 Z1.100 F7800.000
@Object AIl_In_One_3D_printer_test_micro.stl id:0 copy 0
```

in cancel object settings, i've set, as suggested in the readme, the following regex:
```
; process (.*)
;MESH:(.*)
; printing object (.*)
```

none of these will catch the gcode created by Slic3r PE.
Any hint ?

paukstelis, 2019-03-29:
> 
> I'm trying to use this plugin with Slic3r PE, latest beta.
> 
> In the GCODE I can see:
> 
> ; stop printing object AIl_In_One_3D_printer_test_micro.stl id:0 copy 0
> ;M117 Printing layer 4
> ;LAYER:4
> G1 Z1.100 F7800.000
> @Object AIl_In_One_3D_printer_test_micro.stl id:0 copy 0
If you see the ‘@Object’ line that means it is working in detecting objects. What happens when you load the gcode? You should then see the objects in the cancel tab.

guestisp, 2019-03-29:
Yes, I can see them, probably I had a temporary issue yesterday.
But I don't see any object in the gcode visualizer. I've thought that was possible to cancel object by choosing them also in gcode visualizer

closed 2020-04-04 by paukstelis