paukstelis/Octoprint-Cancelobject #31: Canceling first model freezes print

Atomsk102, 2019-04-15:
If I cancel the first model in the list, then at the beginning of the next layer, the printer freezes.  The head doesn't move, and the heaters stay on.

This may be a conflict with Octolapse, which takes control of the printer at the beginning of each layer, to move the head out of the way, for a snapshot.

paukstelis, 2019-04-15:
What is in the terminal when it freezes?

Atomsk102, 2019-04-15:
Terminal contents in attached text file.  I've also included the Gcode file that was running at the time.
[CE3_Cube Test.gcode.txt](https://github.com/paukstelis/Octoprint-Cancelobject/files/3082304/CE3_Cube.Test.gcode.txt)
[Terminal Dump.txt](https://github.com/paukstelis/Octoprint-Cancelobject/files/3082306/Terminal.Dump.txt)

When I try to cancel the print, The terminal says:

Changing monitoring state from "Printing" to "Cancelling"[...]

Communication timeout while printing, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.

paukstelis, 2019-04-15:
Switch to relative extrusion in your slicer. It simplifies things greatly. I have absolute extrusion working OK on my printer, but its something that I've seen enough inconsistencies to not recommend it.

You are right it probably is related to OctoLapse. Can you try disabling OctoLapse and trying again?

closed 2019-08-31 by paukstelis