paukstelis/Octoprint-Cancelobject #32: Wrong object cancelled

paulr4, 2019-04-16:
Hi, I noticed this on a print containing 72 copies of a small object. (image attached)

I clicked the centre object; it was the first to print and had not adhered to the bed properly. Immediately a 'cancel object' box popped up, and I confirmed. The object up and left from my chosen object greyed out, and the printer skipped this object in subsequent layers.

I was only able to cancel my intended object by clicking down and to the right from its actual location, which leads me to think there's some problem with the way the plugin is inferring the object from the click location.

[image]

I'm running OctoPrint 1.3.10 on a Raspberry Pi using OctoPi.

paukstelis, 2019-04-16:
What slicer are you using? 

If you could provide the gcode that would be useful.

paulr4, 2019-04-16:
No problem, here's the file. It's from Slic3r PE's latest beta release, 1.42.0-beta2.

[Cable clamps (72).zip](https://github.com/paukstelis/Octoprint-Cancelobject/files/3086952/Cable.clamps.72.zip)

Clicking 'Refresh objects' doesn't seem to help with the mouse selection problem. The points where a mouse click causes the cancel popup to appear do not correspond with the orange object markers. The markers themselves are in exactly the right place over each object, as you can see in the image above.

My laptop is running at 125% scale, I wonder if this is the problem? Browser is Chrome.

paukstelis, 2019-04-16:
I don't see any issue when I run your gcode with a virtual printer in Octoprint.
I do recommend switching to using relative extrusion distances.
Something to try:
Reinstall the plugin. The newest version provides the object name in the confirmation dialogue:
[image]
Now you can cancel from the tab and verify that the right object gets cancelled in the gcode:
[image]
[image]

paulr4, 2019-04-16:
Ok thanks, I'll try that tomorrow, it's running a long print right now.

paukstelis, 2019-04-17:
Thanks for figuring that out. I'm not sure why that causes a problem or if there is a fix for that. I don't have a Windows computer so it would be difficult to test and debug.

paulr4, 2019-04-17:
I see, that will be difficult. Perhaps someone else can suggest something. Another little piece of the puzzle is that when switching scaling, the correct/incorrect clicking behaviour remains until I reload OctoPrint in the browser. Only then does the change become apparent.

I've changed the name of this issue to better match the problem, and I'll happy to give feedback if you want to try a fix. Thank you very much for your work on this plugin, I'm already finding it useful.

paulr4, 2019-04-17:
I just had a little delve into your code and found the part in cancelobject.js that handles clicks on the canvas. Line 296 is the following:

`var pixRatio = window.devicepixRatio || 1;`

I can't find many references to 'devicepixRatio' online; Google redirects me to 'devicepixelRatio'. Could it be that this is a mis-spelling, and that as a result pixRatio defaults to 1 regardless of the real ratio?

paukstelis, 2019-04-17:
Hmm...quite possible. Good catch.

paukstelis, 2019-04-18:
If you could, please reinstall from URL here: https://github.com/paukstelis/Octoprint-Cancelobject/archive/0.3.2.zip
and test again.

paulr4, 2019-04-18:
Thanks, please forgive my ignorance but how do I do that? I have only installed plugins through the plugin manager before.

paukstelis, 2019-04-18:
No worries. Use the Plugin Manager ->Get More->Install from URL

paukstelis, 2019-04-22:
Any chance to test this? I'll make a new release if this fixes the problem.

paulr4, 2019-04-23:
Sorry, I've been away for much of the Easter weekend, I hope to try it in the next 24h...

paulr4, 2019-04-30:
Sorry for the delay, I've tried the update and saw no orange dots at all, I'm not sure why. Changing back to the current release I see no dots either. I'm confused and still trying to get to the bottom of it. Wondering if I perhaps have another plugin that's interfering.

paulr4, 2019-04-30:
I've been tinkering some more and gained a little more understanding.

The reason for my previous post is that my installation is very slow to analyse a new file. I'm printing large files with a lot of individual parts, and only using a Pi Zero W. There's a delay of a minute of so between loading the file and seeing the parts list appear under the Cancel Objects tab, and if I don't wait patiently enough, it acts as if there are no labelled parts in the file at all.

My original problem of clicks being registered in the wrong place seems unaffected by your update, sadly. So I guess we are back to square one. Perhaps some debugging info about where clicks are being registered might help to diagnose the problem?

paukstelis, 2019-04-30:
I quickly abandoned my attempts at using a Pi Zero W with octoprint. Just not stable enough.

It actually does log the x,y position of the click to the browser console so you can open that up to see where it thinks it is clicking. If you tried that with and without the window scaling (by perhaps clicking between two objects) it might be instructive.

paulr4, 2019-05-01:
I think I found the problem, it's devicePixelRatio not devicepixelRatio; my mistake. Fixing that in the code running on my machine makes everything work fine.

paukstelis, 2019-05-02:
Great. Thanks for looking into this. I will make the correction and push a new release in the next couple of days.

closed 2019-05-02 by paukstelis