paukstelis/Octoprint-Cancelobject #33: Canceling Last Object Nullifies Ending Gcode

Budge72, 2019-04-19:
I love this plugin.  But recently, I cancelled the last object on my buildplate (the last object to print on each layer).  At the end of the very last layer, the print head stopped on the object before the cancelled one, and froze.  Octoprint no longer had any control to move the print head, turn off the heaters, or stop the print.  I had to do all these things using the LCD control on my printer.  But the hot end was left on for a couple of hours right on my PETG object at 240C.  I guess by cancelling the last object, Octoprint never got to run the ending gcode.

paukstelis, 2019-04-19:
Glad you like it. This can be avoided. Look at the instructions. Basically, just add `; process ENDGCODE` at the start of your ending gcode script in your slicer.

Budge72, 2019-04-20:
; process ENDGCODE   or   ;MESH:ENDGCODE    if I'm using Cura?

paukstelis, 2019-04-20:
> ; process ENDGCODE or ;MESH:ENDGCODE if I'm using Cura?

Either will work.

closed 2019-08-31 by paukstelis