paukstelis/Octoprint-Cancelobject #34: File section

buliwyf42, 2019-06-04:
When i install the cancel Objects Plugin the files Section on the left gehts out of shape.
This is how it looks like if i disable the plugin: 
[image]
And this is how it looks like if i enable it. 
[image]
I have the Version 0.3.1

paukstelis, 2019-06-04:
I don't see that behavior, but I'll look into it. What other plugins do you have installed?

buliwyf42, 2019-06-04:
Several, but i disabled nearly all of them to figure it out. I just let some originals plugins from foosel enabled.

paukstelis, 2019-06-05:
Can you install from here: https://github.com/paukstelis/Octoprint-Cancelobject/archive/0.3.2.zip and see if this makes any difference?

buliwyf42, 2019-06-05:
This looks definitely better
better. 
[image]

paukstelis, 2019-06-05:
OK. I will put this out along with some other bug fixes in a new release. Thank you for reporting.

closed 2019-06-05 by paukstelis