paukstelis/Octoprint-Cancelobject #35: Upload to octoprint "watched" directory?

bluegizmo83, 2019-06-25:
"If you upload files directly to your octoprint instance, direct them to the watched directory and not the upload directory."

How exactly is this done?? I can't find any information about how to direct all uploads to the Watched DIR in octoprint... I can see that a DIR named "Watched" exists, but I can't figure out how I'm suppose to get my files to upload to there when I hit "Print with Octoprint" in CURA 4.0

paukstelis, 2019-06-25:
It should work fine without modification using the Cura Octoprint Plugin. You can confirm by uploading a file to octoprint from Cura and seeing if the objects get loaded in the CancelObject tab.

closed 2019-08-31 by paukstelis

Haprog, 2025-04-05:
What about with PrusaSlicer? Should it work when I "upload" or "upload & print" directly from PrusaSlicer?

Would still be nice to know when and how to use the watched directory.