paukstelis/Octoprint-Cancelobject #37: Minor NavBar annoyance probably easy to fix

PRGeno, 2019-08-12:
I have the "Show active in NavBar" option unchecked because I have other information shown in the NavBar and adding the object information overflows the NavBar and makes a mess of the page.

That setting almost works as expected, except for one minor issue which occurs when a print ends.  There must be a statement to refresh the NavBar with the phrase "Current Object: None" when the print ends, and this shows up in the NavBar regardless of the "Show active in NavBar" setting.  This phrase is pretty useless and unfortunately stays there until the page is manually refreshed.

Is there a way this can be removed, as it really has no use, or at least is made to honor the "Show active in NavBar" setting?

paukstelis, 2019-08-13:
Thanks. Will look into it.

paukstelis, 2019-08-13:
Fix pushed, but untested. Please reinstall and give it a try.

PRGeno, 2019-08-13:
After a quick test it appears to have worked as expected.

Thank you for the incredibly quick response

closed 2019-08-31 by paukstelis