paukstelis/Octoprint-Cancelobject #41: Plugin crashes when printing multicolor with a purge tower

gdombiak, 2020-01-04:
I finally tracked this one down. I'm using PrusaSlicer and diving into MultiColor printing. I then ran into this issue: https://github.com/tomvandeneede/p2pp/issues/54. After a lot of analyzing and narrowing down the issue I got to the root of it.

The purge tower created by PrusaSlicer uses gcode like this:
> G1 Y140.344 E1.85090

When this happens, the following error is triggered

>2020-01-04 17:20:17,345 - octoprint.util.comm - ERROR - Error while processing hook cancelobject for phase queuing and command G1 Y140.344 E1.85090:
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python3.8/site-packages/octoprint/util/comm.py", line 3333, in _process_command_phase
    hook_results = hook(self, phase, command, command_type, gcode, subcode=subcode, tags=tags)
  File "/home/pi/oprint/local/lib/python3.8/site-packages/octoprint_cancelobject/__init__.py", line 480, in check_queue
    if e_move[0] > obj["max_x"]:
TypeError: '>' not supported between instances of 'NoneType' and 'float'

There are 2 things that I think could be improved:
1. Add a try/catch so that any exception in the plugin is trapped and logged. Right now they are not trapped so OctoPrint **DOES NOT** send the gcode to the printer. This is what was so hard to narrow down what was going on.
1. Add logic to detect None in X or Y during extrusion movements
1. Maybe add logic to detect what are purge towers and treat them differently in the plugin

I can create a PR for the first 2 things if that would help. Ignoring purge towers or treat them as another object is more than I can take. :)

Thanks,
Gaston

paukstelis, 2020-01-05:
I think option 1 is probably most straightforward. The code is really pretty junky there. A think a helper function is in order for those tracking object. It can just go through the list of e_move and update. The try/except can live in that. Feel free to PR it,, or I can get to it in a few days. I think that would also be a good time to do a new release so people will get the bugfixes/improvements you've got in so far. Maybe I can do a release without messing it up for a change :)

gdombiak, 2020-01-05:
PR created. Implemented suggestions 1 and 2.

closed 2020-01-05 by gdombiak