paukstelis/Octoprint-Cancelobject #45: after update 0.4.0 upload from Prusa Slicer error

cypeks, 2020-01-11:
When I try to send gcode to Octoprint from Prusa Slicer, I get the message:

```
Error sending to print server:
HTTP 500: <! DOCTYPE HTML PUBLIC "- // W3C // DTD HTML 3.2 Final // EN">
<title> 500 Internal Server Error </title>
<h1> Internal Server Error </h1>
<p> The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application. </p>
```

log from octoprint:
```
2020-01-11 19:55:12,297 - octoprint - ERROR - Exception on /api/files/local [POST]
Traceback (most recent call last):
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1276, in decorated_view
    return no_firstrun_access(flask_login.login_required(func))(*args, **kwargs)
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1295, in decorated_view
    return func(*args, **kwargs)
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/Flask_Login-0.2.11-py2.7.egg/flask_login.py", line 758, in decorated_view
    return func(*args, **kwargs)
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/server/api/files.py", line 361, in uploadGcodeFile
    display=canonFilename)
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/filemanager/__init__.py", line 493, in add_file
    path_in_storage = self._storage(destination).add_file(path, file_object, links=links, printer_profile=printer_profile, allow_overwrite=allow_overwrite, display=display)
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/filemanager/storage.py", line 765, in add_file
    file_object.save(file_path)
  File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/filemanager/util.py", line 92, in save
    shutil.copyfileobj(source, dest)
  File "/usr/lib/python2.7/shutil.py", line 52, in copyfileobj
    fdst.write(buf)
UnicodeEncodeError: 'ascii' codec can't encode character u'\u0142' in position 479: ordinal not in range(128)
2020-01-11 19:55:12,301 - tornado.access - ERROR - 500 POST /api/files/local (192.168.1.221) 249.60ms
```

Octoprint: 1.3.12
Prusa Slicer 2.1.1 linux64

After disabling the plugin "Cancel Objects 0.4.0", Octoprint works normally.

paukstelis, 2020-01-11:
Uninstall/reinstall the plugin. The issue has been recognized and fixed.

cypeks, 2020-01-12:
It works. Thank you.

closed 2020-01-12 by cypeks