paukstelis/Octoprint-Cancelobject #46: Canceling object spams gcode

NovaViper, 2020-01-13:
Hey. I added in some gcode that whenever an object gets canceled, it makes the printer raise up by 10mm to avoid hitting any other object that remains on the bed. However, when I go and cancel one object, it seems to continuously spam the gcode I have set, which causes my printer to keep on raising the gantry until it smacks into my frame. I'm running Octoprint 1.3.12 on OctoPi 0.16.0 with the latest version of CancelObjects. Could you please fix this?
This is the gcode here.
[image]

And this is the terminal showing the spamming
[image]

paukstelis, 2020-01-13:
I will look into this. For clarification, where are you readjusting the Z-height after you lift?

paukstelis, 2020-01-13:
If you can, uninstall and reinstall from this URL to test: https://github.com/paukstelis/Octoprint-Cancelobject/archive/0.4.1.zip

NovaViper, 2020-01-13:
@paukstelis It still spams the gcode. And it doesn't install 0.4.1 but 0.4.0 instead

paukstelis, 2020-01-13:
Yes, it is still tagged as 0.4.0. It doesn't spam any gcode in my tests. Perhaps try removing your M85s and see what happens.

paukstelis, 2020-01-13:
What printer/firmware? What hardware are you using to run OctoPrint?  I've tested with the same commands you have and I don't see an issue. Can you attach your gcode?

NovaViper, 2020-01-13:
Creality Ender 3 with Marlin 2.0 on the SKR V1.3

⁣Get BlueMail for Android ​

NovaViper, 2020-01-13:
For running Octoprint, I'm using the raspberry pi 3B+

And the screen shot I sent is the gcode in using for the command 

⁣Get BlueMail for Android ​

NovaViper, 2020-01-13:
Here's the gcode

NovaViper, 2020-01-13:
Here's the part im testing it on. https://gist.github.com/NovaViper/bd4c97bb50bbafee92e336a58d8ad05e

And I went and reinstalled the plugin twice, but it still keeps spamming the gcode

paukstelis, 2020-01-13:
I am fairly certain this is a firmware compatibility issue. One of these commands is probably a queue-jumper in Marlin 2.0 that leads to the loop. See what happens when you remove all M85s from before and after gcode.

NovaViper, 2020-01-13:
Nope, didn't fix it

paukstelis, 2020-01-13:
It won't happen in the terminal because of how the command buffer works. This certainly doesn't reproduce on my Prusa. I'll flash the latest Marlin 2.0 on another printer and see what happens.

paukstelis, 2020-01-14:
I flashed latest Marlin 2.0 (bugfix) on a printer and could not reproduce this issue. This leads me to believe that it is not something related to the plugin, but is either firmware or hardware or perhaps some combination. Wish I could tell you more, but I can't fix a problem I can't see.

NovaViper, 2020-01-14:
Idk if it will help any, but I do have Action commands turned on in the firmware. And the marlin 2.0 im using is the full release version (the non-bugfix version)

paukstelis, 2020-01-24:
When you are working with this, are you cancelling just a single object, or are there multiple objects that get cancelled. I continued to test this and the only time I can reproduce with your before gcode is when I have multiple objects and I cancel several in a row.  In that case it does do multiple z lifts since it is hitting multiple cancelled objects.

NovaViper, 2020-01-25:
The very first time, I did cancel multiple objects. I can't remember the second time around though. I'm not at my house to verify this unfortunately

paukstelis, 2020-07-15:
Without more to work with here, I am going to close this for now. Please let me know if you are still experiencing this issue.

closed 2020-07-15 by paukstelis