paukstelis/Octoprint-Cancelobject #47: Latest commit causes 500 internal server error

nigeljohnson73, 2020-01-17:
I have just upgraded to the latest version as OctoPrint reminded me it was available however since then I cannot upload files to the server. Disabling the plugin fixes my issue and the exception seems to be related to code changes in the latest commit. I am sending files via the API from PurusaSlicer 2.1.1 and via dragging files in the web interface. Here is one file I have tried, but it fails on at least 3 different files (everything I have tried to send since the update)

[bit.gcode.zip](https://github.com/paukstelis/Octoprint-Cancelobject/files/4075360/bit.gcode.zip)

Here is the log file entry

```
2020-01-17 06:24:20,542 - octoprint - ERROR - Exception on /api/files/local [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1276, in decorated_view
    return no_firstrun_access(flask_login.login_required(func))(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1295, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask_login.py", line 758, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/files.py", line 361, in uploadGcodeFile
    display=canonFilename)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 546, in new_add_file
    return self._file_manager.original_add_file(destination, path, file_object, links, allow_overwrite, printer_profile, None, display)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/filemanager/__init__.py", line 493, in add_file
    path_in_storage = self._storage(destination).add_file(path, file_object, links=links, printer_profile=printer_profile, allow_overwrite=allow_overwrite, display=display)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/filemanager/storage.py", line 765, in add_file
    file_object.save(file_path)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/filemanager/util.py", line 92, in save
    shutil.copyfileobj(source, dest)
  File "/usr/lib/python2.7/shutil.py", line 49, in copyfileobj
    buf = fsrc.read(length)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/filemanager/util.py", line 194, in read
    processed_line = self.process_line(line)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_cancelobject/__init__.py", line 37, in process_line
    line = self._matchComment(line)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_cancelobject/__init__.py", line 45, in _matchComment
    line = line.decode("utf-8", "strict")
  File "/home/pi/oprint/lib/python2.7/encodings/utf_8.py", line 16, in decode
    return codecs.utf_8_decode(input, errors, True)
UnicodeEncodeError: 'ascii' codec can't encode character u'\u010d' in position 200: ordinal not in range(128)
2020-01-17 06:24:20,554 - tornado.access - ERROR - 500 POST /api/files/local (2a00:23c6:a12b:1d01:f802:b563:2dab:f82f) 5653.48ms
```

paukstelis, 2020-01-17:
Uninstall and reinstall the plugin. There were issues with how it handled unicodes. It has been fixed.

nigeljohnson73, 2020-01-18:
Perfect, thanks :)

closed 2020-01-18 by nigeljohnson73