paukstelis/Octoprint-Cancelobject #50: Canceling objects affects the print quality of nearest/next part.

sgano, 2020-04-03:
I am using PrusaSlicer v2.2.0, on a Prusa MK3S, via OctoPrint 1.4.0 running on OctoPi 0.17.0, and using version 0.4.1.

( I have also enabled "Label objects" and added "; printing object ENDGCODE " to the start of the custom "G-Code" section).

After canceling an object, the print quality of the remaining part(s), typically the nearest part (or the part after the skipped part), is significantly impacted.  I am not sure if it has to do with the z-hop or proper retractions while skipping the canceled object.    But where the next object picks up after a skipped object there is a lot of extra material like oozing and stringing (typically nearest the closest point to the cancelled object and the next part starts).   

(Not sure if this is a slicer issue, a plugin issue, or a setup issue.)

paukstelis, 2020-04-04:
Probably a setup issue. Make sure you use relative extrusion distances.

sgano, 2020-04-06:
Okay, thanks I checked and I have been using relative extrusion distances and seem to have everything else setup properly; but will keep experimenting.

closed 2020-04-06 by sgano