paukstelis/Octoprint-Cancelobject #55: Cancel Objects and Arc Welder don't play nice

shiftingtech, 2020-06-21:
I'm not entirely sure what's happening, but I'm using SuperSlicer, with the object labeling turned on, and the Arc Welder Octoprint plugin:

Cancel objects correctly recognizes the objects in the original GCode file, but not the arcwelder-ized version ("Object list populated when GCODE file is loaded", even though I HAVE loaded the gcode)

As near as I can tell, the object labeling is still present in the arcwelder version, so I'm unsure what's actually causing the issue.  I've attached a sample with both the
 original file, and the arcwelder version.  arcwelder version is the .aw.gcode

[demo_gcode.zip](https://github.com/paukstelis/Octoprint-Cancelobject/files/4810458/demo_gcode.zip)

paukstelis, 2020-06-21:
I believe this is because the `@` commands that are the converted comments are case sensitive. The are `@OBJECT` in the Arc Welder file and `@Object` in the other. Not sure how that happened, but you might do a simple search and replace to see if it works. I will try when I get the chance, but I'm bogged down in other things at the moment.

paukstelis, 2020-06-21:
I'm guessing ArcWelder must uppercase all comments for some reason.

paukstelis, 2020-06-21:
If this is indeed the case, I will make this case insensitive in the next release (which may be a while yet).

shiftingtech, 2020-06-21:
ah! interesting, I totally didn't register that.
I'll test shortly.
(also, if it does turn out to be that, I'll file a bug on arcwelder side to leave the case alone)

shiftingtech, 2020-07-07:
My apologies for not returning to this sooner.  I can confirm that you are correct, and the case change is the issue.

paukstelis, 2020-07-15:
Ignoring case for the object tags will be included in 0.4.2 Closing for now.

closed 2020-07-15 by paukstelis