paukstelis/Octoprint-Cancelobject #57: UnicodeEncodeError: 'ascii' codec can't encode character u'\xb0' in position 19: ordinal not in range(128)

coldtobi, 2020-08-15:
When uploading this gcode file, either manually or with prusa slicer to octoprint 1.4.1, CancelObject fails the upload with the Unicode Error.  Disabling the CancelObject plugin in octoprint will make the upload work again.
The installed version is 0.4.3

As I cant upload the gcode to this issue: You can find it here on my nextcloud instance:
https://sviech.de/s/dksgFerWWTnPd7C/download

(I've tested whether the "°" symbol in the filename is the culprit, but it is not; it seems to be the parsing withing cancelobject.)

Log:
```
Aug 15 12:27:34 bananapi2 octoprint[1409]: 2020-08-15 12:27:34,664 - octoprint.server.util.sockjs - INFO - User _admin logged in on the socket from client 10.x.x.68
Aug 15 12:27:40 bananapi2 octoprint[1409]: 2020-08-15 12:27:40,144 - octoprint - ERROR - Exception on /api/files/local [POST]
Aug 15 12:27:40 bananapi2 octoprint[1409]: Traceback (most recent call last):
Aug 15 12:27:40 bananapi2 octoprint[1409]:   File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/flask/app.py", line 2447, in wsgi_app
Aug 15 12:27:40 bananapi2 octoprint[1409]:     response = self.full_dispatch_request()
Aug 15 12:27:40 bananapi2 octoprint[1409]:   File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/flask/app.py", line 1952, in full_dispatch_request
Aug 15 12:27:40 bananapi2 octoprint[1409]:     rv = self.handle_user_exception(e)
Aug 15 12:27:40 bananapi2 octoprint[1409]:   File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/flask/app.py", line 1821, in handle_user_exception
Aug 15 12:27:40 bananapi2 octoprint[1409]:     reraise(exc_type, exc_value, tb)
Aug 15 12:27:40 bananapi2 octoprint[1409]:   File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/flask/app.py", line 1950, in full_dispatch_request
Aug 15 12:27:40 bananapi2 octoprint[1409]:     rv = self.dispatch_request()
Aug 15 12:27:40 bananapi2 octoprint[1409]:   File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/flask/app.py", line 1936, in dispatch_request
Aug 15 12:27:40 bananapi2 octoprint[1409]:     return self.view_functions[rule.endpoint](**req.view_args)
Aug 15 12:27:40 bananapi2 octoprint[1409]:   File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1315, in decorated_view
Aug 15 12:27:40 bananapi2 octoprint[1409]:     return func(*args, **kwargs)
Aug 15 12:27:40 bananapi2 octoprint[1409]:   File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/vendor/flask_principal.py", line 199, in _decorated
Aug 15 12:27:40 bananapi2 octoprint[1409]:     rv = f(*args, **kw)
Aug 15 12:27:40 bananapi2 octoprint[1409]:   File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/server/api/files.py", line 411, in uploadGcodeFile
Aug 15 12:27:40 bananapi2 octoprint[1409]:     display=canonFilename)
Aug 15 12:27:40 bananapi2 octoprint[1409]:   File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_PrintTimeGenius/__init__.py", line 554, in new_add_file
Aug 15 12:27:40 bananapi2 octoprint[1409]:     return self._file_manager.original_add_file(destination, path, file_object, links, allow_overwrite, printer_profile, None, display)
Aug 15 12:27:40 bananapi2 octoprint[1409]:   File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/filemanager/__init__.py", line 501, in add_file
Aug 15 12:27:40 bananapi2 octoprint[1409]:     path_in_storage = self._storage(destination).add_file(path, file_object, links=links, printer_profile=printer_profile, allow_overwrite=al
low_overwrite, display=display)
Aug 15 12:27:40 bananapi2 octoprint[1409]:   File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/filemanager/storage.py", line 774, in add_file
Aug 15 12:27:40 bananapi2 octoprint[1409]:     file_object.save(file_path)
Aug 15 12:27:40 bananapi2 octoprint[1409]:   File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/filemanager/util.py", line 92, in save
Aug 15 12:27:40 bananapi2 octoprint[1409]:     shutil.copyfileobj(source, dest)
Aug 15 12:27:40 bananapi2 octoprint[1409]:   File "/usr/lib/python2.7/shutil.py", line 63, in copyfileobj
Aug 15 12:27:40 bananapi2 octoprint[1409]:     buf = fsrc.read(length)
Aug 15 12:27:40 bananapi2 octoprint[1409]:   File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/filemanager/util.py", line 198, in read
Aug 15 12:27:40 bananapi2 octoprint[1409]:     processed_line = self.process_line(line)
Aug 15 12:27:40 bananapi2 octoprint[1409]:   File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_cancelobject/__init__.py", line 39, in process_line
Aug 15 12:27:40 bananapi2 octoprint[1409]:     line = self._matchComment(line)
Aug 15 12:27:40 bananapi2 octoprint[1409]:   File "/home/octoprint/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_cancelobject/__init__.py", line 60, in _matchComment
Aug 15 12:27:40 bananapi2 octoprint[1409]:     line = "{0}stop {1}\n".format(self._reptag, stop.group(1))
Aug 15 12:27:40 bananapi2 octoprint[1409]: UnicodeEncodeError: 'ascii' codec can't encode character u'\xb0' in position 19: ordinal not in range(128)
Aug 15 12:27:40 bananapi2 octoprint[1409]: 2020-08-15 12:27:40,546 - tornado.access - ERROR - 500 POST /api/files/local (10.243.48.68) 2406.61ms

```

paukstelis, 2020-08-15:
It is the issue with the symbol in the comment that the plugin uses (which comes from the STL filename):
`; printing object Left-Fan-Nozzle-180°.stl id:0 copy 0`
This had been worked-around previously, but recent changes may have reverted. I will look into fixing.

paukstelis, 2020-08-15:
Fixed. Remove and reinstall plugin to get changes. I'll push 0.4.4 once I fix a few other things.

paukstelis, 2020-08-25:
Closing. Corrected in 0.4.4.

closed 2020-08-25 by paukstelis