paukstelis/Octoprint-Cancelobject #58: How to cancel objects?

rewolff, 2020-09-26:
I have a "plate" of 64 (identical) small objects. Last time I printed that after a while three started to become loose so I installed "cancelobject". By the time the print had finished 10 to 15 of the objects had failed because of the printing-in-the-air  caused stuff to be dragged around and knocked others off. 

So now I tried to use "cancelobject" I just see: 
`Object list populated when GCODE file is loaded
`
but my GCODE has already been loaded. It's printing, I can view the layers in the gcode viewer tab, and by now it's started on the BRIM. 

According to https://github.com/paukstelis/Octoprint-Cancelobject/issues/24 I should have ";process" tags in my gcode. I checked but I don't. How do I get those? < tried to read documentation on my own and... I upgraded to "latest development branch" of slic3r" and hit "export GCODE". Still no tags. 

I just noticed I now have a "cancel objects" line underneath the gcode viewer window. but I don't understand how to cancel objects. The documentation hints at big yelow dots on objects, but I don't get those. 
[image]

rewolff, 2020-09-26:
According to: 
https://www.reddit.com/r/octoprint/comments/bf3emm/be_careful_using_the_cancel_objects_plugin/
I need to turn on "label prints with object id". In the manual 

Arrgh! Missed that in the manual. Sorry about that. Enabled that... and... Still no labels.

rewolff, 2020-09-26:
OK. Found it. I need to hit "split". Of course I don't like arranging the 49-225 objects manually. I prefer them to be arranged by openscad. So I have openscad generate the up to 225 objects in a regular grid and load that as one STL. 

After this finishes printing I'm going to print another test with again "only" 49 of these, to see if I can cancel one. then I'll run the production run of 225 of them.

closed 2020-09-26 by rewolff

rewolff, 2020-09-26:
OK. On the second print, I tried to cancel the middle one while things were heating up. Works! 

Next: What are the buttons below the gcode-viewer supposed to do?: I see nothing happening when I click them.

paukstelis, 2020-09-26:
Each object will have an orange dot in the gcode viewer which you can use to cancel the object. Sometimes prime lines or brims are included in one object so the the position of the dot is incorrect. You can reset the positions  to remove the current positions, then you can refresh the positions which are calculated as each object gets extruded.

killernils, 2020-11-16:
To be honest, those buttons doesnt seem to do anything for me either. I can confirm that your plugin has processed my gcode and I can see the objects in the Cancel-Objects tab.
However, no orange dots in gcode viewer!

rewolff, 2020-11-16:
That's what the "refresh objects" button is for. And with toggle markers you can turn the orange blobs on and off. 

What I find annoying is that I do my normal workflow (design one copy, test that, then tell openscad that I want 250 copies and export that as one STL, slice and send to printer). Now when things have been going well and I haven't needed cancel object for a while, I'll have forgotten to "split" the object in slic3r and.... cancel object doesn't work. :-(

paukstelis, 2020-11-16:
There is no way around that. Object delineation happens at the slicer level.

rewolff, 2020-11-16:
Yeah. It is easy to look at it that way. But that doesn't mean it always has to stay like that. 

Now the objects I was printing that started this start out as one "base" and after a while you get two "towers" and a rim that is less high. Looking at it at that point in time, if you detect three and later two objects and require me to cancel them all individually, that'd be way better than ruining an additional 255 of my 256 objects. 

And providing that option would also create new possibilities that weren't possible earlier. For example, if I'm printing an octopus with its 8 legs in the air.... What if one breaks off partway through the printjob? A user might chose to continue to print the other seven and fix up the missing leg later. Ask the slicer and it's one object. Do a "naive" object detection on the current (or last) layer and you'll get 8 or 9 distinct printed areas that don't touch. Consider that cancel-able objects and you're done.

paukstelis, 2020-11-16:
"Detecting" objects is not so straightforward. It can be parsed, in some cases, from certain slicers, but for the use case you are talking about, ExcludeRegion plugin works great.

killernils, 2020-11-16:
> That's what the "refresh objects" button is for. And with toggle markers you can turn the orange blobs on and off.
>
> What I find annoying is that I do my normal workflow (design one copy, test that, [...]

I may be stupid... But I hit the "refresh object" in gcode viewer mode. Wait one layer of all objects, and tries to toggle markers. Nothing happens. Using latest  version of Chrome.
Tried with/without "sync with job progress" if that should matter?

rewolff, 2022-11-20:
I've had this way too many times.... Can't find the "label objects" option. 

Of course it is in print settings -> output options . 
(That's where I expected it, but I spent half an hour looking through all other places becauase I didn't see it there)....

But you need a recent version of Slic3r.