paukstelis/Octoprint-Cancelobject #62: Slow travel to cancelled object on every layer

yorb, 2020-11-29:
Say I'm printing two parts, A and B, and I cancel part B after a few layers. From then on, on each layer, when it gets to the end of the extrusion for part A, it does this:

1. An extremely slow travel move to where it would have started extrusion for part B (I would guess about 10–15 mm/s)
2. A series of quick travels back to part A (following the lines of where part B would have been)

This takes up time, but more importantly it allows plenty of material to ooze out during the slow travel, causing some unsightly  blobs on part A (and increased risk of catching one and knocking the part).

Yes I'm using relative extrusion, Cura 4.8, Marlin 2.0.x. Is there any way to avoid this? :/

paukstelis, 2020-11-29:
Please provide sample gcode. First guess is that it is related to wipe moves.

yorb, 2020-11-29:
Sure, here's the file I'm printing right now that is showing the issue. I don't have Wipe Between Layers on, but Outer Wall Wipe Distance is set to 0.2 and Infill Wipe Distance is at 0.125 (both Cura defaults, I think).

[SpeakerMountPartB.gcode.zip](https://github.com/paukstelis/Octoprint-Cancelobject/files/5611568/SpeakerMountPartB.gcode.zip)

paukstelis, 2020-11-29:
It is certainly related to the wipes. Here is the first snippet:

```
G0 F3600 X170.223 Y128.221
G1 F1500 X173.726 Y131.724 E0.33984
G1 X173.814 Y131.812
G1 F4200 E-5
@Object SpeakerMountPartA.stl
G0 F3600 X174.435 Y132.707
G0 X174.158 Y134.191
G0 X170.975 Y149.804
G0 X170.374 Y151.187
G0 X168.36 Y167.336
G0 X168.057 Y167.308
;TYPE:WALL-INNER
G1 F4200 E5
G1 F1500 X168.332 Y167.308 E0.01886
G1 X168.331 Y167.033 E0.01219
G1 X168.279 Y157.422 E0.65931
G1 X168.295 Y151.758 E0.38855
G1 X168.528 Y151.446 E0.02671
```
You see that it is doing the G0 moves that are the wipe for the previous object (part B) after the `@` tag for PartA. You can file a bug/request with CuraEngine, and/or remove all the wipes and just change your setup to do infill first.

paukstelis, 2020-11-29:
It looks like there is some new behavior in Cura with respect to the types of movements they are also putting in NONMESH between objects on layer changes that is causing issues as well.

yorb, 2020-11-29:
Oh interesting. Though, it seems like the G0 moves are part of the travel to the other part, so it kinda makes sense that they're after the tag. If I step through the actions in the Preview, it seems like those moves include all the combing moves required to get "out" of the current part and to the starting point in the next part.

So I guess what's happening is those moves to get TO the cancelled part are gone, but the moves to get BACK out of the cancelled part are still there. Doesn't really explain why it goes so slow though, does it?

paukstelis, 2020-11-29:
If you look at the NONMESH object you'll see the problem. There is an F600 move. I believe if you cancel the last object in the list you'll do this F600 to the cancelled object.

paukstelis, 2020-11-29:
It does seem to largely be related to combing, but also some redundant moves in the NONMESH.

yorb, 2020-11-29:
Yep, you're totally right. Thanks for figuring that out so quickly! I might experiment with turning combing off, though it does seem to help quite a bit and I rarely need to cancel objects... but when I do, it sure is a lifesaver. 😅

hayden-t, 2022-06-29:
here is my gcode around the time it happens, there is a move from the back of the bed (high y) to the front (low y) and a z layer change

```
Send: N93073 G0 F15000 X50.41 Y168.792*119
Recv: ok
Send: N93074 G1 F2400 E3477.97688*53
Recv: ok
Recv:  T:200.49 /200.00 B:65.03 /65.00 @:35 B@:33
Send: N93075 G92 E3792.53465*111
Recv: X:50.41 Y:168.79 Z:4.48 E:3792.53 Count X:4033 Y:13503 Z:1771
Recv: ok
Send: N93076 G0 F300 X98.067 Y47.879 Z4.8*43
Recv:  T:200.40 /200.00 B:65.03 /65.00 @:37 B@:33
Recv: echo:busy: processing
Recv:  T:200.38 /200.00 B:65.05 /65.00 @:37 B@:31
Recv: echo:busy: processing
Recv:  T:200.23 /200.00 B:65.02 /65.00 @:39 B@:34
Recv: echo:busy: processing
Recv:  T:200.23 /200.00 B:65.03 /65.00 @:38 B@:32
Recv: echo:busy: processing
Recv:  T:200.21 /200.00 B:65.03 /65.00 @:37 B@:32
Recv: echo:busy: processing
Recv: ok
Send: N93077 G0 F15000 X96.579 Y48.174*122
Recv: ok
Send: N93078 G0 X95.875 Y47.352*44
Recv:  T:200.17 /200.00 B:65.05 /65.00 @:38 B@:30
Recv: ok
Send: N93079 G0 X80.562 Y33.322*38
Recv:  T:200.16 /200.00 B:65.00 /65.00 @:38 B@:34
Recv: echo:busy: processing
Recv:  T:200.17 /200.00 B:65.01 /65.00 @:37 B@:32
Recv: echo:busy: processing
Recv: ok
Send: N93080 G0 X80.35 Y32.26*33
Recv: ok
Send: N93081 G0 X35.154 Y28.978*33
Recv:  T:200.09 /200.00 B:65.04 /65.00 @:39 B@:28
Recv: echo:busy: processing
Recv:  T:200.07 /200.00 B:64.96 /65.00 @:38 B@:34
Recv: echo:busy: processing
Recv:  T:199.99 /200.00 B:64.88 /65.00 @:39 B@:39
Recv: echo:busy: processing
Recv:  T:200.00 /200.00 B:64.90 /65.00 @:39 B@:37
Recv: echo:busy: processing
Recv:  T:200.10 /200.00 B:64.83 /65.00 @:36 B@:43
```

hayden-t, 2022-06-30:
ok ill try that next print, i did find the suspect command in the original gcode too, at the start of a nomesh section 
```
;MESH:NONMESH
G0 F300 X98.067 Y47.879 Z4.8
```

hayden-t, 2022-06-30:
I turned off combing and regenerated the gcode to inspect manually while waiting for print to finish, and it makes no difference it still has the z change line pasted in last comment with 300 speed part of nomesh

paukstelis, 2022-06-30:
I would complain to Cura devs. I've tried to get them to not do stupid things, but it falls on deaf ears. They pretty much only care how it impacts Ultimakers.

paukstelis, 2022-06-30:
PrusaSlicer/SuperSlicer work nicer.

hayden-t, 2022-06-30:
I can look into those. But wouldnt it be possible to just look at the nonmesh section after the cancelled object and strip any x y movement from the z change ?

paukstelis, 2022-06-30:
Possible, sure. I'm not really keen on writing a bunch of exceptions for each individual slicer.

hayden-t, 2022-06-30:
yeah i can understand that as its a bit hacky too, how does this plugin 'cancel' an object ? is the does it intercept the gcode live or modify the loaded gcode in memory or file ?

paukstelis, 2022-06-30:
The only way it can work is to do it "live". OctoPrint has the ability to modify commands before they are sent to the printer. It just watches that queue, and if the object is cancelled it just skips sending those commands to the printer.

paukstelis, 2022-06-30:
I just had a look at Cura 5, and it looks like it fixes this problem.

hayden-t, 2022-06-30:
dang, im on win 7 which doesnt support 5.0

paukstelis, 2022-06-30:
And to clarify, it seems to fix with combing turned off or to infill only. Same behavior if combing is set to all. The other thing that is puzzling about your gcode is how slow that first move is. Even before Cura 5 (I only had 4.11 on hand) it would at least use the travel move speed and not the Z-move speed. So if you can upgrade even a bit the results will be better.

hayden-t, 2022-06-30:
i tried combing off and it didnt change the gcode output order. 
i just installed cura 4.13.1, changed no settings, added to objects at either end of y plate, sliced and its still putting the actual Z change in the NONMESH section before the LAYER section

`G0 X33.29 Y185.742
G1 F1500 E309.04907
G1 X33.163 Y185.615 E309.05205
G0 F9000 X32.976 Y185.841
G1 F1500 E302.55205
;MESH:NONMESH
G0 F300 X32.976 Y185.841 Z3.2
G0 F9000 X31.958 Y184.549
G0 X32.602 Y183.804
G0 X31.977 Y31.986
G0 X31.977 Y31.001
G0 X25.245 Y29.664
G0 X24.424 Y29.315
G0 X23.691 Y28.807
G0 X23.078 Y28.158
G0 X22.61 Y27.399
G0 X22.307 Y26.558
G0 X22.183 Y25.675
G0 X22.241 Y24.784
G0 X22.48 Y23.925
G0 X22.889 Y23.134
G0 X23.465 Y22.427
G0 X24.428 Y21.809
G0 X25.268 Y21.506
G0 X26.154 Y21.378
G0 X26.942 Y21.459
G0 X27.654 Y21.715
G0 X34.425 Y26.371
G0 X34.719 Y26.495
G0 X35.069 Y26.463
G0 X35.361 Y26.273
G0 X35.529 Y25.966
G0 X35.532 Y25.618
G0 X31.354 Y20.679
;TIME_ELAPSED:939.952262
;LAYER:15
;TYPE:WALL-INNER
;MESH:DragonClip.stl
G1 F1500 E309.05205
G1 X31.336 Y20.668 E309.05275
G1 X30.905 Y20.619 E309.06718`

hayden-t, 2022-06-30:
that is stock settings, this is with combing off:

```
G0 F9000 X33.29 Y185.742
G1 F1500 E309.04905
G1 X33.163 Y185.615 E309.05204
G1 F1500 E302.55204
;MESH:NONMESH
G0 F300 X33.163 Y185.615 Z3.2
G0 F9000 X31.354 Y20.679
;TIME_ELAPSED:1064.725393
;LAYER:15
;TYPE:WALL-INNER
;MESH:DragonClip.stl
G1 F1500 E309.05204
G1 X31.336 Y20.668 E309.05274
```

hayden-t, 2022-06-30:
someone discussing a similar problem https://community.ultimaker.com/topic/33422-insert-g-code-before-layer-change-is-at-the-wrong-place/

hayden-t, 2022-06-30:
actually these last 2 posts would be ok, as the z rise is at the original y position (back of plate), will try it out next time

paukstelis, 2022-06-30:
Trying adding Z-hop and see how it looks.

hayden-t, 2022-06-30:
sorry the last few posts i was just looking at which black the z change was, not its y position, so it seems i can get my original cura 4.8 to make gcode with the z height change before xy change to next part, doesnt seem to be combing related, i will investigate further and get back to you

hayden-t, 2022-06-30:
after putting this down and reflecting on it some, I cant see how it can be possible to get around this while the unskippable z layer change has a x and y coord mentioned in it, as there will always be a object skipping scenario where the z change is out somewhere there is no objects anymore and the head inadvertently travels to that x y at the slow z speed to make that z change. 

the only way around this is to strip out the x/y , can you think of any scenario where it would need to be there ?
btw here is photo of my first job with 2 skipped objects (last 2 in series)
[image]

paukstelis, 2022-06-30:
This why I was telling you to take it up with Cura devs. Cura puts in unnecessary moves. The thing you might look into is why that motion is so slow on your profile. F300 is ridiculously slow. In all my profiles that same movement is F10800, so while the behavior is the same, it doesn't string because it is just a very fast travel movement.

hayden-t, 2022-06-30:
So im running Creality Slicer 4.8 which im assuming is just cura 4.8 rebadged with their profiles, as it has a ender 3 S1 Pro profile and came with the printer, if i use the default standard slice profile it still has F300 as the Z speed, if a set a custom FFF printer profile it goes to F600.

I set up a custom printer profile in cura 4.13.1 and it is atleast not setting a Z speed wheres as the ender 3 pro profile i tried earlier (as closest to mine) was still fixed at 300.
4.13.1 custom fff:
```
G0 F7200 X108.867 Y90.765
G1 F600 X108.772 Y90.67 E41.31693
;MESH:NONMESH
G0 X108.772 Y90.67 Z0.9
G0 F7200 X110.75 Y91.267
G0 X110.752 Y91.611
;TIME_ELAPSED:39.256577
;LAYER:3
;TYPE:WALL-INNER
;MESH:MiniBall.stl
```

hayden-t, 2022-06-30:
Im going to have to leave this again for a few days, now i cant stop 4.13.1 from putting in F600 on z change now and cant work out how that happened or how to stop it...

hayden-t, 2022-07-02:
Ok I narrowed in on where the F300 or F600 is coming from in Cura. It is the Z Hop speed !! But regardless of if Z hop is enabled or not and thus the setting is shown or not it is still being used. and it is defaulting to the printer profile for the json setting machine_max_feedrate_z and for my creality based profile it would not let me go above 10mm/s (F600) without it going red and blocking slicing. 

A custom profile however can go above this, but it seems a bug that it would be being used even when z hop is off, I will try a cura github issue to get some clarification.

Still cant work out how I got it to not mention the speed F600 in that layer change....

So another possible work around could be to use the find replace post processing plugin in cura to strip out and F600 mentions, or you could make the z hop speed like F12345 so its quite unique and then find replace it with nothing.

hayden-t, 2022-07-02:
o dear, it was right under our nose all this time 
[image]

hayden-t, 2022-07-02:
https://github.com/Ultimaker/Cura/issues/12658

hayden-t, 2022-07-02:
So it seems the easiest way to get around this for now for me, is actually just set the Z hop speed to 0, i then get F0 in the layer change, and luckily for my printer it doesnt just stop, and travels at what seems to be my normal travel speed ! another option would be to set it to your travel speed if F0 doesnt work for you.

hayden-t, 2022-07-02:
The other option would be to look into this https://github.com/shinmai/cura-M486