paukstelis/Octoprint-Cancelobject #64: Cancel Objects non functional in GCode Viewer tab

kb-, 2021-01-03:
The orange dot isn't displayed on objects in my GCode Viewer tab (after loading object or while printing). Nothing happens when using the 3 buttons ("Toggle markers", "Reset objects", "Refresh objects"):

[image]

Objects are listed in Cancel Objects tab tho:
[image]

Object change is detected while printing:
[image]

The GCode has the "printing object" tags:
```
; printing object Socket_flange.stl id:1 copy 0
...
; stop printing object Socket_flange.stl id:1 copy 0
; printing object locking_nut.stl id:0 copy 0
```

[fan_holder-4.zip](https://github.com/paukstelis/Octoprint-Cancelobject/files/5761799/fan_holder-4.zip)

GCode is generated by PrusaSlicer 2.2.0, with the options as listed in the comments at the end of the GCode file.
Cancel Objects 0.4.4

An error shows up in the log:
_

> 2021-01-03 04:33:32,486 - octoprint.plugin.core - ERROR - Error loading plugin cancelobject
> Traceback (most recent call last):
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 1298, in _import_plugin
>     module = _load_module(module_name, spec)
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 69, in _load_module
>     return imp.load_module(name, f, filename, details)
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_cancelobject/__init__.py", line 13, in <module>
>     from flask.ext.login import current_user
> ImportError: No module named ext.login

_
[log.zip](https://github.com/paukstelis/Octoprint-Cancelobject/files/5761835/log.zip)

browser.user_agent: Mozilla/5.0 (Windows NT 6.3; Win64; x64; rv:85.0) Gecko/20100101 Firefox/85.0
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1400
env.hardware.ram: 918188032
env.os.bits: 32
env.os.id: linux
env.os.platform: linux2
env.plugins.pi_support.model: Raspberry Pi 3 Model B Plus Rev 1.3
env.plugins.pi_support.octopi_version: 0.15.1
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 9.0.3
env.python.version: 2.7.13
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.5.2
printer.firmware: Marlin V1; Sprinter/grbl mashup for gen6

paukstelis, 2021-01-03:
For some reason you do not have the latest version of the plugin. The error that is reported was something that fixed back in July. Have you tried uninstalling the plugin and reinstalling?

kb-, 2021-01-03:
I did that. The plugin manager offered me 0.4.4. 
At the beginning of the posted log I had 0.4.1; at the end 0.4.4.

paukstelis, 2021-01-13:
I have not been able to reproduce this. My first thought was that there was something wrong with absolute extrusion and tracking, but it works fine for me. Could you try relative extrusion and see if it makes any difference?

paukstelis, 2021-01-13:
Your gcode works fine for me, so there must be something else (plugin?) that is causing an issue.

[image]

tablatronix, 2021-04-10:
Doesn't work for me either. I just uninstalled and reinstalled. The 3 buttons do not appear to do anything..

paukstelis, 2021-04-10:
First place to look will be in the web browser console. Just to confirm, when you try to use the buttons you do have visible objects that have orange dots on them?

tablatronix, 2021-04-10:
no dots, no change at all for either of them, checked the console no errors or anything.

paukstelis, 2021-04-10:
What other plugins do you have installed?

paukstelis, 2021-04-10:
FYI, there shouldn't be any dots before printing (unless you use SuperSlicer). The dots are done by tracking where extrusions are taking place during the printing of an object.

the0bone, 2022-03-08:
I just had a similar effect. Printed a Design from Fusion and while I'm with a german version, Fusion names to bodies as "Körper".
The special character ö caused the PlugIn to not understand the different objects.
The Status of current object was always no and the dots did not show up.

After changing the name all is fine.