paukstelis/Octoprint-Cancelobject #66: event to other plugin

giloser, 2021-02-12:
Hi thanks for the plugin I really love it.

I work on the telegram plugin and someone ask if it could be possible to send a message when we changed object.
I see that you send one for the frontend, would it be possible also to fire an event for the python plugin part?

something like eventManager().fire("CancelObject_update", info of the object)

thanks a lot

paukstelis, 2021-03-10:
There is now a change_event branch that will trigger on object updates and object cancellations.If you can give that a try and report back how it works, I will push it in the next release.

giloser, 2021-03-12:
thanks a lot!
I'll try that as soon as I can.

paukstelis, 2021-04-01:
Get a chance to check this? I'm going to put out 0.4.5 shortly and will include the changes if they work for your application.