paukstelis/Octoprint-Cancelobject #68: No orange marker and 10 minutes pauses for each skipped blocs

PrimAvalou, 2021-05-18:
Running OctoPrint 1.6.1 / CancelObject 0.4.4 / Raspberry Pi 3B+ 1GB

I'm running DIY 3D Printers for years, but very new to use OctoPrint. I installed the plugin a few days ago, but I always get the same issue cancelling an object : 
I sliced the following gcode file using Slic3r : [210518_CubeCalibration_Size0,75_x2.gcode.txt](https://github.com/paukstelis/Octoprint-Cancelobject/files/6502441/210518_CubeCalibration_Size0.75_x2.gcode.txt)
Running that gcode file, I cancelled the object on the right side while the one on the left side is being printed. At the end of the left object layer, the printer run the Before skip GCODE (G91,G1 Z10 F400) and then stop for 10 minutes. Then printer executes the After skip GCODE (G91,G1 Z-10,G90), the extruder run to the right backward corner of the plate (I don't why) and resume left object printing.
Here is a CancelObject log file, my last print test began at 16:36:47 : [plugin_cancelobject_cancelobject.log.txt](https://github.com/paukstelis/Octoprint-Cancelobject/files/6502630/plugin_cancelobject_cancelobject.log.txt)

By the way, on the GCODE Viewer, I don't see any orange point. Maybe it is a different issue, I could create another Issue discussion.

**Plugins installed :** 
themeify
simpleemergencystop
gpiocontrol
fullscreen
floatingnavbar
cancelobject
octolapse
marlingcodedocumentation
consolidatedtabs
prettygcode

Thanks for your help !

PrimAvalou, 2021-05-18:
In addition, when the 10 minutes pause occurs I get a "Stabilization error" message from OctoPrint. Is my Raspberry Pi struggling ?

paukstelis, 2021-05-18:
In no scenario should there be a 10 minute pause. I would first try without the Before/After gcode. In most cases, that isn't needed. I will do some tests with your provided gcode.

Have you hit the Refresh button after there have been some extrusions to see the orange dot? The plugin tracks object positions so it has to go through some extrusions first before it will appear.

PrimAvalou, 2021-05-18:
Yeah, I firstly tried without Before/After gcodes, I added them to understand where does the print stopped, and to easily explain what happened too. Let me know what results you obtain with my gcode, I just add that I always launch the print under the Octolapse test mode (for me it doesn't change anything as it doesn't change the gcode file itself, just ignoring temps lines and extruder commands).

For orange markers, yes I tried to hit the Reset as well as the Toggle buttons, without success. But I always cancelled a body from the first layer, so I'll try to wait some layers.

paukstelis, 2021-05-18:
I have heard that there are issues with Octolapse and CancelObjects (though I haven't seen them myself) so it wouldn't be a bad idea to disable Octolapse and use something like DryRun plugin instead. I saw no issues with your gcode on my setup. Everything worked as expected.

PrimAvalou, 2021-05-18:
I confirm that it works without any pause using the DryRun plugin, thanks !! 

But I still don't see orange markers on the GCode viewer, even waiting for about 3 layers. Since my first post, I installed the UI Customizer plugin, that shows a little window on the GCode viewer on the right sidebar. But I don't think that is the cause actually, because I waited at least for a complete layer during during my very first tests. I can still use the CancelObjects tab to cancel the desired object, but that should be nice to get orange markers too.

I'll try tomorrow for a real print, without any test mode or the DryRun plugin. Do you know if Octolapse could interfere with CancelObject ? Or is it just its test mode that creates issues ?

paukstelis, 2021-05-18:
That's good to know. It has been a long time since I used Octolapse. Back 3 or so years ago it did not cause any issues. I'll have to fire it up and check again.

PrimAvalou, 2021-05-21:
OK, so next to some prints (some was to big to be shown in the GCODE viewer, that's why I answer so late), and after disabling Octolapse, the plugin works well to cancel the print of an object. No any 10 minutes pause, or things like that. Using the CancelObject tab, the plugin works perfectly for me. 

But when I could use the GCODE viewer tab, I've never seen any orange marker...

paukstelis, 2021-05-21:
1. Confirm, you have pushed the Refresh button after you open gcode viewer?
2. What browser are you using? Are there any errors in the browser console log?

PrimAvalou, 2021-06-04:
Hi !  Sorry for not responding for weeks. I mounted a new printers, and all my prints since was too big for being loaded in the GCODE viewer. 

I recently could print a small object, and I effectively saw the orange markers next to push on the Reset button. There was only one object, so could not valid markers work perfectly well, but I think my next prints will be too large to load them in the GCODE viewer. My main issue was from the Octolapse interaction, that has been solved.

So many thanks, and I close this issue.

closed 2021-06-04 by PrimAvalou