paukstelis/Octoprint-Cancelobject #71: Update to 4.0.7 failed : "zipfile.BadZipFile: File is not a zip file"

taikolito, 2022-02-16:
octoprint 1.7.3

- click "proceed" in octoprint's plugin update pop-in
- the update window shows the following messages :

`Updating, please wait.
+++++++++++++++++++++++++++++++++++
Now updating Cancel object to 0.4.7
+++++++++++++++++++++++++++++++++++
/home/pi/oprint/bin/python3 -m pip --disable-pip-version-check install https://github.com/paukstelis/OctoPrint-Cancelobject/archive/0.4.7.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/paukstelis/OctoPrint-Cancelobject/archive/0.4.7.zip
Downloading https://github.com/paukstelis/OctoPrint-Cancelobject/archive/0.4.7.zip (105 bytes)
ERROR: Exception:
Traceback (most recent call last):
File "/home/pi/oprint/lib/python3.7/site-packages/pip/_internal/cli/base_command.py", line 224, in _main
status = self.run(options, args)
File "/home/pi/oprint/lib/python3.7/site-packages/pip/_internal/cli/req_command.py", line 180, in wrapper
return func(self, options, args)
File "/home/pi/oprint/lib/python3.7/site-packages/pip/_internal/commands/install.py", line 321, in run
reqs, check_supported_wheels=not options.target_dir
File "/home/pi/oprint/lib/python3.7/site-packages/pip/_internal/resolution/resolvelib/resolver.py", line 122, in resolve
requirements, max_rounds=try_to_avoid_resolution_too_deep,
File "/home/pi/oprint/lib/python3.7/site-packages/pip/_vendor/resolvelib/resolvers.py", line 445, in resolve
state = resolution.resolve(requirements, max_rounds=max_rounds)
File "/home/pi/oprint/lib/python3.7/site-packages/pip/_vendor/resolvelib/resolvers.py", line 310, in resolve
name, crit = self._merge_into_criterion(r, parent=None)
File "/home/pi/oprint/lib/python3.7/site-packages/pip/_vendor/resolvelib/resolvers.py", line 169, in _merge_into_criterion
name = self._p.identify(requirement)
File "/home/pi/oprint/lib/python3.7/site-packages/pip/_internal/resolution/resolvelib/provider.py", line 60, in identify
return dependency.name
File "/home/pi/oprint/lib/python3.7/site-packages/pip/_internal/resolution/resolvelib/requirements.py", line 41, in name
return self.candidate.name
File "/home/pi/oprint/lib/python3.7/site-packages/pip/_internal/resolution/resolvelib/candidates.py", line 188, in name
return self.project_name
File "/home/pi/oprint/lib/python3.7/site-packages/pip/_internal/resolution/resolvelib/candidates.py", line 182, in project_name
self._name = canonicalize_name(self.dist.project_name)
File "/home/pi/oprint/lib/python3.7/site-packages/pip/_internal/resolution/resolvelib/candidates.py", line 239, in dist
self._prepare()
File "/home/pi/oprint/lib/python3.7/site-packages/pip/_internal/resolution/resolvelib/candidates.py", line 226, in _prepare
dist = self._prepare_distribution()
File "/home/pi/oprint/lib/python3.7/site-packages/pip/_internal/resolution/resolvelib/candidates.py", line 319, in _prepare_distribution
self._ireq, parallel_builds=True,
File "/home/pi/oprint/lib/python3.7/site-packages/pip/_internal/operations/prepare.py", line 480, in prepare_linked_requirement
return self._prepare_linked_requirement(req, parallel_builds)
File "/home/pi/oprint/lib/python3.7/site-packages/pip/_internal/operations/prepare.py", line 505, in _prepare_linked_requirement
self.download_dir, hashes,
File "/home/pi/oprint/lib/python3.7/site-packages/pip/_internal/operations/prepare.py", line 263, in unpack_url
unpack_file(file.path, location, file.content_type)
File "/home/pi/oprint/lib/python3.7/site-packages/pip/_internal/utils/unpacking.py", line 261, in unpack_file
flatten=not filename.endswith('.whl')
File "/home/pi/oprint/lib/python3.7/site-packages/pip/_internal/utils/unpacking.py", line 129, in unzip_file
zip = zipfile.ZipFile(zipfp, allowZip64=True)
File "/usr/lib/python3.7/zipfile.py", line 1222, in __init__
self._RealGetContents()
File "/usr/lib/python3.7/zipfile.py", line 1289, in _RealGetContents
raise BadZipFile("File is not a zip file")
zipfile.BadZipFile: File is not a zip file
The update did not finish successfully. Please consult <code>octoprint.log</code> and <code>plugin_softwareupdate_console.log</code> for details.
`

see octoprint.log here : [octoprint-12.log](https://github.com/paukstelis/Octoprint-Cancelobject/files/8079789/octoprint-12.log)
see plugin_softwareupdate_console.log here : [plugin_softwareupdate_console.log](https://github.com/paukstelis/Octoprint-Cancelobject/files/8079802/plugin_softwareupdate_console.log)

taikolito, 2022-02-16:
edit : title

paukstelis, 2022-02-16:
Should be fixed now. closing.

closed 2022-02-16 by paukstelis

taikolito, 2022-02-16:
I confirm, update ok now
Thanks