paukstelis/Octoprint-Cancelobject #72: Objects are not canceled

Revilote, 2022-03-09:
Hello,
When i click "cancel" the button disappears for a moment, then apperas below the objects name and then appears at the normal position again. The object is not deleted and will still be printed..
What do i do wrong?

I am using Prusaslicer with a clone MK3S.
Rasperry Pi 3B with newest Octoprint.

The Picture is when the button is pushed.
[image]

paukstelis, 2022-03-09:
Do you get a confirmation window when you click cancel?

Revilote, 2022-03-09:
No. Nothing happens except the short buttons position change.

paukstelis, 2022-03-09:
Perhaps it is an incompatible plugin. What version of OctoPrint do you have installed and what plugins are you using?

Revilote, 2022-03-10:
I'm using version 1.7.3.
Plugins are those in the picture:
[image]

paukstelis, 2022-03-10:
Please post contents of plugin_cancelobject_cancelobject.log, octoprint.log (after you attempt to cancel), and any errors that appear in your web browser console after trying to cancel.

Revilote, 2022-03-10:
Unfortunately i dont know how to get those files...

Revilote, 2022-03-17:
Okay. I found it.
[image]

Revilote, 2022-04-04:
So what can i do to solve this problem?
It´s not just this file, but all my files. The names are given by prusaslicer.

Aron3DP, 2022-07-27:
I have the same issue. Tried to do it two different times and no confirmation ever pops up. The cancel button moves below the name of the object and then back to the side.

paukstelis, 2022-07-27:
Provide your gcode file and a list of plugins you are using.

Aron3DP, 2022-07-27:
Here is the gcode file and a screen of all the plugins used.
[electronics-set_0.2mm_240C_PETG_MINI_5h49m.zip](https://github.com/paukstelis/Octoprint-Cancelobject/files/9201375/electronics-set_0.2mm_240C_PETG_MINI_5h49m.zip)

paukstelis, 2022-07-27:
It works fine for me, even with all the plugins installed (with the exception of OctoLapse). Perhaps there is some incompatibility with the display properties in the browser. 

When you click the cancel button, can you open up the development tools/inspectr  in the browser (typically ctrl+shift+I) and see if there is anything that looks like this? Or if there are errors in the console?
[image]

paukstelis, 2022-07-27:
What theme are you using for UI customizer?

Aron3DP, 2022-07-27:
I'll check the other when I am home later and can test. But Theme is the Prusa Tribute theme or what ever it's called.

paukstelis, 2022-07-27:
That theme is the problem. As soon as I switched to that I see the same behavior.

Aron3DP, 2022-07-27:
I would have never thought to check that. It did seem like a UI issue with how the cancel button would move under the object name and back.

paukstelis, 2022-07-27:
The theme overwrites the CSS for the button padding, which moves the button. A work-around to still use the theme is to to click and hold, then move the cursor to the button position and release. I will see if there is anything that I can do on my side, but I think   it might have to be adjusted in the theme.

Aron3DP, 2022-07-27:
Alright thank you. I could use another theme as well in the mean time. What theme for sure works? I am not really tied to the theme.

paukstelis, 2022-07-27:
That's the only one that I have seen that has a problem, though I did not go through and try them all. An additional alternative is that you can cancel objects in the gcode viewer as well. Just click refresh objects after each has done a bit of printing and you will see orange dots corresponding to each object.

Aron3DP, 2022-07-27:
Ok, will do. Thanks for the fast replies.