paukstelis/Octoprint-Cancelobject #73: excessive stringing after canceling object

rchamp26, 2022-05-12:
Hi there, 

been doing some testing on this plugin and have noticed that after cancelling the object, the plugin does indeed cancel it (though the cancel button in the plugin view, doesn't grey out as i've seen in youtube demo videos). 

However, it seems the retraction from the object before to the object after the cancelled object does not occur. I assume this is expected behaviour as the Gcode to retract to move to the cancelled object and then the gcode to retract when it goes from cancelled object to the object after is no longer present and skipped. 

my typical retraction is 6.5mm with .2mm zhop 40mm/s retraction/deretraction. Should i enter gcode in the before skip or after skip section of the plugin settings? can you assist with what the gcode should look like? i'm running an anycubic vyper with community marlin firmware and prusa slicer for the slicer

rchamp26, 2022-05-12:
G1 X48.431 Y194.255 E.47063 ; infill
; stop printing object rr-head.stl id:5 copy 0
; printing object rr-bady-r.stl id:2 copy 0
G1 E-7 F2400 ; retract
G1 Z.45 F9000 ; lift Z
G1 X58.139 Y195.287 ; move to first perimeter point
G1 X58.139 Y201.393 ; move to first perimeter point
G1 X131.183 Y203.527 ; move to first perimeter point
G1 X135.715 Y203.529 ; move to first perimeter point
G1 X147.757 Y204.811 ; move to first perimeter point
G1 Z.25 ; restore layer Z
G1 E7 F2400 ;  ; unretract
;TYPE:Perimeter
;WIDTH:0.58
G1 F3150

then the next section
G1 X182.596 Y206.956 E.06331 ; infill
; stop printing object rr-bady-r.stl id:2 copy 0
; printing object rr-bady-l.stl id:1 copy 0
G1 E-7 F2400 ; retract
G1 Z.45 F9000 ; lift Z
G1 X146.517 Y173.65 ; move to first perimeter point
G1 Z.25 ; restore layer Z
G1 E7 F2400 ;  ; unretract
;TYPE:Perimeter
;WIDTH:0.58
G1 F3150

paukstelis, 2022-05-16:
This is likely because the retracts are within the next object, and not at the end of the previous object. I don't recall if this is normal, or something new in PS.

rchamp26, 2022-05-21:
Looks like adding in the before and after cancel object gcode that i extracted from my original gcode seems to do the job correctly. Might need to tweak and for some reason the move speed from the before skipped object ot the after skipped object seems significantly slower than all my other travel moves. might need to manually tell it to move faster but the retraction seems to handle the majority of the stringing i was previously getting

closed 2022-05-21 by rchamp26