paukstelis/Octoprint-Cancelobject #76: Feature Request: Klipper EXCLUDE_OBJECT support

KidCe, 2023-03-26:
Hi,
i really love the plugin functionallity but i have some troubles when using it with octoprint and klipper.

But since Klipper now supports the exclude objects itself it would be great if you could implement the needed commands for this.

This would mean CancelObject plugin doesnt need to do any gcode removal / editing, just sending the right command:
EXCLUDE_OBJECT [NAME=object_name]
as Gcode output to klipper. Klipper will handle all the processing itself.

While the Fuildd and Mainsail webinterfaces for klipper already support this feature, for octoprint there is currently no plugin that just forwards the needed command based on clicks on the gcode viewer. Since click on object on gcode viewer is already implemented, I would guess that adding the klipper functionallity would be an easy addition :)

 From a quick search i found out what command fluidd internally uses to set a object as excluded: 
https://github.com/fluidd-core/fluidd/blob/ff50e72b17089f702b23f212d4895c4e57e5d7f1/src/components/widgets/exclude-objects/ExcludeObjectsDialog.vue#L76

From the Klipper documentation there also seems not much that has to be considered:
https://www.klipper3d.org/Exclude_Object.html

I hope that you will still be able to take a look at it, even if you don't use Klipper on your own machines :)

paukstelis, 2023-03-26:
What kind of trouble do you have using with Klipper? It works just fine for me.

This is not something that is within the scope of this plugin, since it has to work for all systems. As of now, as far as I know the current implementation works with Klipper so there is no need to change.

KidCe, 2023-03-26:
just a small problem like the printhead goingover to a already canceled object, but then stopping midway and going back to the actually next object. 

i would imagine a "klipper mode switch" in the plugin, which would then not change any behaviour for other systems. 

If you are not interested in implementing that, maybe you could give me a little guidance in what i would have to look into to make those changed myself on a fork?

paukstelis, 2023-03-27:
> just a small problem like the printhead goingover to a already canceled object, but then stopping midway and going back to the actually next object. 
> 
That has nothing to do with the plugin. That is a slicer issue and how it places object start/stop commands.

> i would imagine a "klipper mode switch" in the plugin, which would then not change any behaviour for other systems. 
> 
> If you are not interested in implementing that, maybe you could give me a little guidance in what i would have to look into to make those changed myself on a fork?

It makes more sense as its own plugin.

tsk-2222, 2024-05-23:
I tried to use PrusaSlicer and Klipper with limited success.

When I export a gcode from Prusaslicer using "Label Objects-Octoprint comments" then things seem to work as expected. However if I use "Label Objects-Firmware specific" then it will label objects for Klipper firmware in this format:
    EXCLUDE_OBJECT_START NAME='awesome.stl'

I tried to add the regex:  EXCLUDE_OBJECT_START NAME='([^']+)'    but that didn't work. 
Is there something else I can do or just keep using Octoprint comments?