paukstelis/Octoprint-Cancelobject #83: Dot in Gcode viewer not alighned in onbject and completely missing in another Browser.

andyspoo, 2024-08-16:
Just to confirm that I am using the latest version of your plugin.
I'm also using Linux (Mint Linux).
The Gcode viewer has been 'refreshed'.
When I view the objects in the gcode viewer Firefox does not show any yellow dots at all.
When I view the same using either the Brave web-browser that is based on Chrome or using Opera, some of the dots are perfect, another is completely missing and another is not on an object.
In the image attached, the large box if just the bounding box.
There are 6 objects/models.
Any ideas what is causing this?
[image]

paukstelis, 2024-08-16:
This is due to how the slicer defines the objects. The bounding box/brim is being considered in object. If you reset the objects using the button, then refresh after it has moved to each object they should be in the right spots.

No idea about firefox.

andyspoo, 2024-08-16:
> This is due to how the slicer defines the objects. The bounding box/brim is being considered in object. If you reset the objects using the button, then refresh after it has moved to each object they should be in the right spots.
> 
> No idea about firefox.

Wow....seriously fast reply!!
I'll give it a go later.
Thanks for the reply.

andyspoo, 2024-08-16:
I'm just starting a new print with the same 6 objects and have just noticed your code is only picking 5 of the 6.
Could this be an Orca Slicer Issue?

[image]

paukstelis, 2024-08-16:
It has to be an Orca Silcer issue. If the correct comments are there, they will get picked up.

paukstelis, 2024-08-16:
If you provide the original gcode file it should be fairly clear what is going on.

andyspoo, 2024-08-16:
Github will not allow me to attach the gcode file. I have uploaded it here. Hopefully it works = 
https://drive.google.com/file/d/14DqT6rR9vdyF26BzOmOHhH-scmyGXTDa/view?usp=sharing

paukstelis, 2024-08-17:
There are only 5 objects tagged in that gcode file. You can look through it and find the `; printing object....` lines. In that case, Orca Silcer must not be tagging one of the objects. Perhaps the one that has the crazy long id number is actually two objects?

andyspoo, 2024-08-17:
I importer 1 .stl file and then 'cloned' it 5 times to make 6. I suspect the one with the long filename might be the original import.
It might be interesting to try importing a model 6 times rather than cloning 1 and see if that makes a difference.
I got Firefox to show the dots BTW, by adding an exception to the Octoprints IP address. It wasn't blocking any other Octoprint code, just yours which seems weird to me. The exceptions I added were for the popup blocker and 'warn when websites try to install ad-ons'.  Firefox wasn't warning of anything, so still strange.

andyspoo, 2024-08-17:
Importing individual stl files made no difference.
Think I need to leave a bug report on the Orca Git.

andyspoo, 2024-08-18:
I have just noticed if I import different models it works fine.
If I either import the same model multiple times or import 1 model and copy it then I have the problem above.
Weird Orca Issue.

andyspoo, 2024-08-18:
Orca bug report here = https://github.com/SoftFever/OrcaSlicer/issues/6494