paukstelis/Octoprint-Cancelobject #84: Gcode uploading file analysis to long when plugin activated

n0dr4c, 2025-01-30:
Hi,

When I activate this plugin, Gcode uploading freeze at process end (when progression bar is full and goes in "saving" state) during nearly 3-4 minutes until it become visible in file list. 
If I disable plugin file appears instantly after upload.
I see no specific logs in plugin or octoprint file.

RPI3B
OctoPrint 1.10.3
Python 3.9.2
OctoPi* 1.0.0cam

12Wakin34, 2025-08-27:
I second that, even 3mb files are slow at "saving" on the card (the upload is normal, but the "save" state is slowed down a lot.
At first I thought it was because of my Pi zero 2W, but when I disable the cancel object plugins, everything is fine.

I got log that says : 
2025-08-27 09:48:14,622 The simple API of this plugin is relying on the default implementation of is_api_protected (newly added in OctoPrint 1.11.2), which in a future version will be switched from False to True for security reasons. Plugin authors should ensure they explicitly declare the API protection status in their SimpleApiPlugin mixin implementation. 

I can't verify if this issue is occuring on Octoprint < 1.11.2

paukstelis, 2025-08-29:
The plugin must process the gcode. The only solution for this is to use better hardware, or to just accept that it takes some time.